micropython/micropython-esp32 #1: Make fails

mwalimu, 2016-12-25:
make fails with this error:

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c
/home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c:16:30: fatal error: bootloader_flash.h: No such file or directory
 #include <bootloader_flash.h>
                              ^
compilation terminated.
Makefile:593: recipe for target 'build/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o' failed
make: *** [build/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o] Error 1

dpgeorge, 2016-12-26:
Please run `make V=1` to see exactly what include paths are passed to gcc, and also report the git tag of the ESP-IDF that you are using.  If you run `make idf-version` it will tell you which ESP-IDF tag MicroPython is known to build with and you might find more success building against that version.

mwalimu, 2016-12-29:
Here is the V=1 output:
 "make V=1
CC /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c
~/esp/crosstool-NG/builds/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DESP_PLATFORM  -I../lib -I../lib/netutils -I. -I.. -I../lib/mp-readline -I../lib/netutils -I../lib/timeutils -Ibuild -I~/esp/esp-idf/components/bootloader_support/include -I~/esp/esp-idf/components/driver/include -I~/esp/esp-idf/components/driver/include/driver -I~/esp/esp-idf/components/nghttp/port/include -I~/esp/esp-idf/components/nghttp/include -I~/esp/esp-idf/components/esp32/include -I~/esp/esp-idf/components/ethernet/include -I~/esp/esp-idf/components/expat/include/expat -I~/esp/esp-idf/components/expat/port/include -I~/esp/esp-idf/components/json/include -I~/esp/esp-idf/components/json/port/include -I~/esp/esp-idf/components/log/include -I~/esp/esp-idf/components/newlib/include -I~/esp/esp-idf/components/nvs_flash/include -I~/esp/esp-idf/components/freertos/include -I~/esp/esp-idf/components/freertos/include/freertos -I~/esp/esp-idf/components/tcpip_adapter/include -I~/esp/esp-idf/components/lwip/include/lwip -I~/esp/esp-idf/components/lwip/include/lwip/port -I~/esp/esp-idf/components/lwip/include/lwip/posix -I~/esp/esp-idf/components/mbedtls/include -I~/esp/esp-idf/components/mbedtls/port/include -I~/esp/esp-idf/components/spi_flash/include -I~/esp/esp-idf/components/vfs/include -I~/esp/esp-idf/components/newlib/platform_include -I~/esp/esp-idf/components/xtensa-debug-module/include -I~/esp/esp-idf/components/wpa_supplicant/include -I~/esp/esp-idf/components/wpa_supplicant/port/include -I~/esp/esp-idf/components/ethernet/include -I~/esp/esp-idf/components/tcpip_adapter/include -DBOOTLOADER_BUILD=1 -I~/esp/esp-idf/components/bootloader_support/include_priv -I~/esp/esp-idf/components/bootloader_support/include -I~/esp/esp-idf/components/micro-ecc/micro-ecc -Wno-error=format -c -MD -o build/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c
/home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c:16:30: fatal error: bootloader_flash.h: No such file or directory
 #include <bootloader_flash.h>
                              ^
compilation terminated.
Makefile:605: recipe for target 'build/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o' failed
make: *** [build/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o] Error 1"

dpgeorge, 2016-12-29:
Please use $(HOME) instead of `~` in your makefile.

mwalimu, 2016-12-29:
Did the trick.  Thanks

dpgeorge, 2016-12-29:
Great!  I added a comment about this in the README, see df98a5c22495e2e2392fb0c707ef83603dbb004e.

closed 2016-12-29 by dpgeorge