micropython/micropython-esp32 #9: Fails to start after install

mkarliner, 2017-01-04:
I've been through all the steps to compile and install successfully.
I've also successfully installed example apps from the IDE examples dir.
When micropython is started, it just prints out

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3ffc0008,len:4
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

continuously. Any suggestions? Board is Olimex core-dev

mkarliner, 2017-01-04:
Changed flashmode to dio and it worked!

closed 2017-01-04 by mkarliner

dpgeorge, 2017-01-05:
Great!  I added a small comment about dio to the README, see a20977a492b7f332440ec986fd0cecb83acac36b.