micropython/micropython-esp32 #11: Missing #define CONFIG_

nubcore, 2017-01-11:
compiler error is encountered with the use of an undefined value for CONFIG_ESP32_WIFI_RX_BUFFER_NUM

This is a recent change to the esp-idf.

Should make menuconfig be used to set this value, or just statically added with the default to sdkconfig.h?

Work around, edit sdkconfig.h and add the following, acceptable values are 2-25 with a default of 25
<pre>
#define CONFIG_ESP32_WIFI_RX_BUFFER_NUM 25
</pre>

nickzoic, 2017-01-11:
Even weirder, it was building for me but crashing out when I went to 'import network', with the error message
```
wifi: init rx buffer number=1073535232
OSError: Wifi Invalid Argument
```
Your #12 has fixed the problem so I've merged that and closed this.

nickzoic, 2017-01-11:
(Actually, I'm less convinced than I was that these problems are related, because my *other* build system still has that problem with your patched version.  Trying to work out what the difference is, so I'll reopen so I don't forget)

dpgeorge, 2017-01-12:
Remember to do "git submodule update --init" after pulling a new version.

nickzoic, 2017-02-28:
Yeah, this was a version mismatch thing, rebuilding everything from scratch fixed it so closing.

closed 2017-02-28 by nickzoic