micropython/micropython-esp32 #18: 8-bit addres I2C

wixa, 2017-01-26:
Hy guys. Anybody faced with the problem of the communication device with 8 bit address? Have a led driver ISSI31FL3733 (http://www.issi.com/WW/pdf/31FL3733.pdf) wich have diferent 8-bit addreses for write and read. When i use i2c.scan() i see my driver on address [80] (1010000), but for write some configure registers i must send to addres [160] (1010000 0) with LSB "0" bit. If i try send data to [160] (10100000) i get "OSError: [Errno 19] ENODEV". Can anybody help me resolve this problam? 

https://gist.github.com/wixa/433cef8e77204600cd6c8982872e8f9d

wixa, 2017-01-26:
Ssory. All work. My misstakes in my code, need seaching.

dpgeorge, 2017-01-30:
I2C addresses are 7-bits long (at least the standard ones are).  The "8th bit" is used to indicate a read/write transaction and is added automatically by the I2C methods.  So just pass the 7-bit address and it should work.

closed 2017-01-30 by dpgeorge