micropython/micropython-esp32 #19: How can i cahnge i2c frequency (speed)

wixa, 2017-01-28:
Can anybody help me? How can i cahnge i2c frequency (speed)

dpgeorge, 2017-01-30:
Please see the docs here: http://docs.micropython.org/en/latest/esp8266/library/machine.I2C.html

You can set the frequency using the constructor, eg: `machine.I2C(scl=scl, sda=sda, freq=400000)`.

wixa, 2017-01-31:
Parametr freq dont work. I trying change it, and see with saleae logic analyzer that Frequince dont change, and its about 200 kHz.

dpgeorge, 2017-01-31:
Did you try decreasing it to, say 50kHz?  It's a software implementation and may not be able to go above 200kHz.

wixa, 2017-01-31:
Ou, yes i try decreasing to 100kHz, and having the same 200kHz on out. It is now for me not important, start  issue only for attention that "freq=" dont work.

dpgeorge, 2017-02-06:
In 38d94222e0eccd04a31b34b4e769fced12811d48 I improved the speed of I2C and SPI.  You should be able to now configure the I2C frequency with more accuracy.

closed 2017-05-24 by dpgeorge