micropython/micropython-esp32 #20: help wanted

wixa, 2017-02-01:
Please can anybody help me understanding. 
1. Is there only one way to upload file script in system using REPL and commands f=open('file.py', 'w' ) and f.write('string....') ? or there are more advanced methods available?

2. In ESP8266 nodeMCU was loasers like ESPlorer or LuaLoader. May be who knows same loaders for ESP32?

3. May be who knows program wich can parsing micropython scripts placing tabulation, transitions to the new line, and special mirrors charset value such as quote character or backslash?

4. I dont know, can i writte such questions in issue in this repository?

Sorry for my questions i am new on github comunity, and still not understand it well the principles and orders.

Thank you for your answers.

nickzoic, 2017-02-01:
Hi, I've been working on this at https://github.com/nickzoic/mpy-utils/
... pretty primitive so far but I hope to improve them as I go along.
There's an uploader script and a FUSE mount for Linux, both using REPL
commands to write files to the filesystem.

-----Nick

--

Nick Moore <nick@zoic.org> 0409 656 267

dpgeorge, 2017-02-02:
To download/upload scripts you can use rshell: https://github.com/dhylands/rshell

To load firmware you can use esptool.py (included as part of the ESP IDF).

The best place to ask such questions is at the forum: https://forum.micropython.org/ .  There's currently no dedicated section for esp32 but there should be soon.

wixa, 2017-02-02:
Thank you guys. Will be good when section for esp32 micropython add.

nickzoic, 2017-02-28:
Yeah, I'm also making a start on more instructions at https://github.com/nickzoic/mpy-tut/ which will grow to cover this stuff.  I've assigned Damien to remind him to make an ESP32 forum!

mattytrentini, 2017-04-12:
Heya Damion, how's creating that ESP32 forum going? ;)

If I can help, please let me know...

mattytrentini, 2017-06-25:
Torwag has created an [ESP32 subforum](https://forum.micropython.org/viewforum.php?f=18). 

@wixa I suggest that if your questions are still open that you re-ask them there. See you in the forum... :)

dpgeorge, 2017-06-26:
Thanks @torwag for creating the subforum.  I've added a picture to distinguish it.

closed 2017-06-26 by dpgeorge