micropython/micropython-esp32 #22: simple example implement own function in micropython

wixa, 2017-02-12:
Hi guys. Can anybody show simple example how implement own function in micropython? As example can be function "turn_led(12)" wich will be set io12 on high level. Or may be another example, but with using hardware.

Thank you all in advance for helping :)

dpgeorge, 2017-02-13:
Please see the documentation for the esp8266, which is very similar: http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html

wixa, 2017-02-13:
Ssory i cant find it. May be you dont understand me. I want implement my own function on "C" level in micropython firmware, and call it from micropython.

dpgeorge, 2017-02-13:
> I want implement my own function on "C" level in micropython firmware, and call it from micropython.

Sorry but I can't provide help/support for this.  Please follow the existing C code and check out the forum at https://forum.micropython.org

closed 2017-02-13 by dpgeorge

nickzoic, 2017-02-14:
Hi Wixa, you can find a brief introduction to wrapping C IDF functions
in my LCA 2017 talk which is online:
http://nick.zoic.org/etc/linuxconf-2017-hobart/[1] ... in summary,
it's a matter of wrapping C objects into Python objects, and
unwrapping them again.

However, most of the time what you want to do to implement higher-level
functionality is to wrap the builtin modules with your own Python
modules .. it's easier and more readable!

cheers,

-----Nick

--

Nick Moore <nick@zoic.org> 0409 656 267

Links:

  1. http://nick.zoic.org/lca2017/

wixa, 2017-02-14:
@dpgeorge Thank you will be seaching more :)

@nickzoic Thank you very interesting lection. Much cleared. Now i need to try :)