micropython/micropython-esp32 #24: Missing machine.enable_irq() etc . .

annejan, 2017-02-19:
Unfortunately micropython-esp32 is missing hardware interrupt routine handling functionality. 

Missing `machine.enable_irq()`, `machine.disable_irq()` and `Pin.irq()` methods.

dpgeorge, 2017-02-20:
Yes it is!  These will be added soon enough.  (See #17 for code that can be used to implement enable/disable irq.)

nickzoic, 2017-02-20:
> Yes it is!  These will be added soon enough.  (See #17[1] for code
> thread[3].

Yeah, it's very high on my list right now ... I'd started on it as part
of #23 but decided to split it out into its own thing.
Thanks for putting it on the list!  

-----N

MrSurly, 2017-05-10:
Both #17 and #23 are marked as closed -- is this issue still needed?

dpgeorge, 2017-05-11:
machine.enable_irq(), machine.disable_irq() and Pin.irq() methods are now all implemented

closed 2017-05-11 by dpgeorge

ril3y, 2019-01-15:
I see this as closed but I cannot seem to find any documentation on using a TouchPad with an IRQ vs polling.  Am I missing something obvious?

MrSurly, 2019-01-15:
This repo is deprecated ESP32 code was pulled into mainline quite some time ago.  [Docs for Pin](http://docs.micropython.org/en/v1.9.3/wipy/library/machine.Pin.html)