micropython/micropython-esp32 #26: OSError: ENOBUFS when instantiating socket object

cversek, 2017-02-22:
Hi, I just got my [WROOM32 dev. board from Adafruit](https://www.adafruit.com/product/3269), following [Tony D's guide](https://www.youtube.com/watch?v=qa2406iiSbI) I built the firmware and got the REPL working - though I had to hold only the boot button when doing `make erase` and `make deploy`.  When trying out the `socket` module I get the following exception:
```
MicroPython v1.8.6-623-g534aefd on 2017-02-22; ESP32 module with ESP32
Type "help()" for more information.
>>> import socket
>>> s = socket.socket()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 105] ENOBUFS
```

cversek, 2017-02-22:
Shortly after, I figured out that this exception does not occur if the `network` module is imported first.  So I guess it's not a big problem, just a bit surprising for a newcomer:
```
>>> import network
I (113597) wifi: frc2_timer_task_hdl:3ffc9f70, prio:22, stack:2048
I (113597) wifi: Init lldesc rx mblock:25
I (113597) wifi: Init lldesc rx ampdu len mblock:7
I (113597) wifi: Init lldesc rx ampdu entry mblock:4
I (113607) wifi: pp_task_hdl : 3ffde14c, prio:23, stack:8192
I (113607) wifi: mode : null
>>> import socket
>>> s = socket.socket()
>>> 

```

nickzoic, 2017-02-22:
Right, yeah, that's an interesting one.  At the moment, 'import network' causes the IDF tcpip_adapter_init function to get called, I think without that you get the problem you described trying to create a socket.
My thinking was that we didn't want to initialize the radio subsystem unless we actually had to, but maybe we need to do tcpip_adapter_init from socket instead ... anyway, good pickup, I'll check it out.

nickzoic, 2017-02-28:
Given that tcpip_adapter_init() includes an explicit `tcpip_inited` flag in case it is run twice, I've put in a PR to just run it when modsocket loads and avoid this confusion: https://github.com/micropython/micropython-esp32/pull/36

dpgeorge, 2017-03-06:
Should now be fixed.

closed 2017-03-06 by dpgeorge