micropython/micropython-esp32 #27: OSError: EAGAIN when socket timedout and other inconsistencies

cversek, 2017-02-23:
Not sure if this is a bug or just a difference between the ESP32 and ESP8266 platforms, but I have this test code `test_socket_timeout.py` which shows the discrepancy:
```python
import network #needed currently for ESP32 socket instantiation
import socket
import errno

SOCKET_TIMEOUT = 10.0

sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.settimeout(SOCKET_TIMEOUT)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
sock.bind(("0.0.0.0", 80))
sock.listen(5)

while True:
    client_sock = None
    try:
        print("Waiting for connection...")
        client_sock, client_address = sock.accept()
        print("accepted connection from '%s'" % (client_address,))
    except OSError as exc:
        if exc.args[0] == errno.ETIMEDOUT:    #case for ESP8266
            print("Caught OSError: ETIMEDOUT")
        if exc.args[0] == errno.EAGAIN:       #case for ESP32
            print("Caught OSError: EAGAIN")
    finally:
        if client_sock:
            client_sock.close()
```
The output for ESP8266:
```
MicroPython v1.8.6-619-g5e15d81-dirty on 2017-02-21; ESP module with ESP8266
Type "help()" for more information.
>>> import test_socket
Waiting for connection...
Caught OSError: ETIMEDOUT
Waiting for connection...
```
But for the ESP32 I get this output:
```
MicroPython v1.8.6-623-g534aefd on 2017-02-23; ESP32 module with ESP32
Type "help()" for more information.
>>> import test_socket
Waiting for connection...
Caught OSError: EAGAIN
Waiting for connection...
```
In either case the socket continues to be able to accept connections the next time (tested with other code);  but, there must be something else going on with the ESP32 because `<ctrl-c>` (`KeyboardInterrupt`) will not break out of the `sock.accept()` call but it will on the ESP8266.

dpgeorge, 2017-02-23:
Note that the ESP32 does not automatically connect to a network.  You need to do that explicitly.

So the fact that you need to "import network" in your script above means that you are not connected to the wifi.  And hence you are getting different errors for socket.accept().

Re ctrl-C not working on ESP32: yes that's a know bug.

cversek, 2017-02-23:
So I beefed up my test code taking into consideration your comments:
```python
import network #needed currently for ESP32 socket instantiation
import socket
import errno

SOCKET_TIMEOUT = 10.0
MY_ESSID = "xxx"
MY_PW    = "xxxxxxxxxx"

sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.connect(MY_ESSID,MY_PW)

while not sta_if.isconnected():
    print("\twaiting for WLAN to connect")
    time.sleep(1.0)

ifconfig = sta_if.ifconfig()
print(ifconfig)

host = ifconfig[0]
addr = socket.getaddrinfo(host,80)[0][-1]

sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.settimeout(SOCKET_TIMEOUT)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
print("binding socket to addr:",addr)
sock.bind(addr)
sock.listen(5)

while True:
    client_sock = None
    try:
        print("Waiting for connection...")
        client_sock, client_address = sock.accept()
        print("accepted connection from '%s'" % (client_address,))
    except OSError as exc:
        if exc.args[0] == errno.ETIMEDOUT:    #case for ESP8266
            print("Caught OSError: ETIMEDOUT")
        elif exc.args[0] == errno.EAGAIN:     #case for ESP32
            print("Caught OSError: EAGAIN")
        else:
            raise
    finally:
        if client_sock:
            client_sock.close()
```
And the socket is still throwing `EAGAIN` exceptions:
```
MicroPython v1.8.6-623-g534aefd on 2017-02-23; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import test_socket
('192.168.1.172', '255.255.255.0', '192.168.1.1', '192.168.1.1')
binding socket to addr: ('192.168.1.172', 80)
Waiting for connection...
Caught OSError: EAGAIN
Waiting for connection...
```
As I have mentioned in passing, I have tested the socket timeouts with other code.   That code actually successfully serves a webpage on the LAN and I was seeing the `EAGAIN` there, so I cooked up this simplified test.

cversek, 2017-02-23:
Reading up on `EAGAIN` in various sources, it seems like a typical response for timeout on a socket accept, the implication is that it is not fatal like `ETIMEDOUT` might suggest.  So maybe its better to leave it alone. I will close the issue a chalk it up to a lesson learned.

nickzoic, 2017-02-23:
> thread[2].

Yeah, that's typically what 'read' would return on a non-blocking read.
From the Linux man page for read(2):

ERRORS

       EAGAIN The  file  descriptor  fd refers to a file other than a
       socket and has been marked non‐
              blocking (O_NONBLOCK), and the read would block.  See
              open(2) for  further  details  on
              the O_NONBLOCK flag.

       EAGAIN or EWOULDBLOCK

              The file descriptor fd refers to a socket and has been
              marked nonblocking (O_NONBLOCK),
              and the read would block.  POSIX.1-2001 allows either
              error to  be  returned  for  this
              case, and does not require these constants to have the
              same value, so a portable appli‐
              cation should check for both possibilities.

However, in this case it isn't a non-blocking read as such, it's a
timing out read :-).  So returning ETIMEDOUT makes more sense,
perhaps.  My feeling is that a) all the Micropythons should do the
same thing as each other and b) that should generally be the same
thing as Python 3 does.

The KeyboardInterrupt exception should be it's own issue ... I just
added it as #28 so we don't forget about it!

-----N

dpgeorge, 2017-02-23:
Looking into the details of lwip on esp32 (which is what we rely on for the socket module), it seems that it's returning EWOULDBLOCK (which equals EAGAIN) when socket.accept() has a timeout.    And it seems that this is indeed the correct behaviour according to POSIX.

But the way CPython (and esp8266 uPy) work is that they have special handling to return ETIMEDOUT in case of a timeout.  In the case of CPython it has a loop that polls the socket for activity, and returns a timeout error if there was no activity within the timeout.  Only if there is activity eg from using select) does it actually call the underlying accept function.  This is how we need to do it with the esp32 port.

dpgeorge, 2017-02-23:
> My feeling is that a) all the Micropythons should do the same thing as each other and b) that should generally be the same thing as Python 3 does.

Yes, so we should fix esp32 :)

> The KeyboardInterrupt exception should be it's own issue

This would also be fixed if we implemented the above idea, of polling the socket for activity.  Because during the poll-loop we would also check for ctrl-C.

nickzoic, 2017-02-28:
Ah, just reread this and it is in `accept` not `read` like I'd assumed. Mind you, the same problem most likely occurs in `read`.  I'm having a look at this now, moving to having sockets always nonblocking and having modsocket handle the apparent blocking (thus interrupts will work too #28)

nickzoic, 2017-02-28:
(making some progress in https://github.com/nickzoic/micropython-esp32/tree/esp32-bugfix-27-socket-accept-eagain but it is still messy and isn't quite there yet)

dpgeorge, 2017-03-08:
The issues here should all be fixed: ctrl-C now works and accept returns ETIMEDOUT.

closed 2017-03-08 by dpgeorge