micropython/micropython-esp32 #29: make of esp32 failed

pici, 2017-02-24:
i followed the setup guide.
cd to esp32 and called make. make is aborted with following message:

> 
CC /Users/Micha/esp/esp-idf/components/bootloader/src/main/bootloader_start.c
LINK build/bootloader.elf
build/bootloader//Users/Micha/esp/esp-idf/components/bootloader/src/main/bootloader_start.o:(.literal.bootloader_main+0x78): undefined reference to `rtc_set_cpu_freq'
build/bootloader//Users/Micha/esp/esp-idf/components/bootloader/src/main/bootloader_start.o: In function `bootloader_main':
bootloader_start.c:(.text.bootloader_main+0x6): undefined reference to `rtc_set_cpu_freq'
bootloader_start.c:(.text.bootloader_main+0xf): undefined reference to `rtc_set_cpu_freq'
collect2: error: ld returned 1 exit status
make: *** [build/bootloader.elf] Error 1
> 
looks like rtc_set_cpu_freq module is missing.

Any hints ?
BRGDS Mike

dpgeorge, 2017-02-25:
rtc_set_cpu_freq is in /esp-idf/components/esp32/lib/librtc.a, which is included in the Makefile.  Did you do "git submodule update --init" in the ESP IDF directory?

JasonBoyles, 2017-02-25:
`git submodule update --init` in the esp-idf directory fixed this problem for me. I cloned esp-idf using `git clone --recursive https://github.com/espressif/esp-idf.git`, but apparently this is not sufficient to update the submodules.

pici, 2017-02-25:
Same solution for me.
Many thanks.

closed 2017-02-25 by pici