micropython/micropython-esp32 #30: unable to build on OSX latest version

diginfo, 2017-02-27:
I have followed the instructions here: http://esp-idf.readthedocs.io/en/latest/macos-setup.html to setup the toolchain and the instructions on this repo to set the path etc, here is the step-by step:

	## Install xtensa
	mkdir -p ~/esp
	cd ~/esp
	## > download xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz
	tar -xzf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz
	
	## Added following to ~/.bash_profile
	export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
	alias get_esp32="export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin"
	
	## source it
	source ~/.bash_profile
	get_esp32
	
	## get ESP-IDF
	cd ~/esp
	git clone --recursive https://github.com/espressif/esp-idf.git
	
	## get sub-modules
	cd ~/esp/esp-idf
	git submodule update --init
	
	## Clone ESP-32 Micropython
	cd ~/esp
	git clone https://github.com/micropython/micropython-esp32.git
	
	## Fix ESP-IDF Version
	cd ~/esp/esp-idf
	git checkout 0a678ebe8c8bb78053f2a2dc4bd26124adde9122
	
	## Create GNUmakefile
	cd ~/esp/micropython-esp32/esp32/GNUmakefile
	
	ESPIDF=/Users/pac/esp/xtensa-esp32-elf/bin
	PORT=/dev/tty.usbserial-DQ008YJN
	FLASH_MODE=qio
	FLASH_SIZE=4MB
	
	include Makefile
	
	## Pre-Compile in Root
	cd ~/esp/micropython-esp32
	make -C mpy-cross
	
	## BUILD
	cd ~/esp/micropython-esp32/esp32
	make
	
After entering the esp-32 folder and running make, it errors at the same point related to bootloader files.

	sudo make
	Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
	fatal: Not a git repository (or any of the parent directories): .git
	** WARNING **
	The git hash of ESP IDF does not match the supported version
	The build may complete and the firmware may work but it is not guaranteed
	ESP IDF path:       /Users/pac/esp/xtensa-esp32-elf/bin
	Current git hash:   
	Supported git hash: 0a678ebe8c8bb78053f2a2dc4bd26124adde9122
	fatal: Not a git repository (or any of the parent directories): .git
	LINK build/bootloader.elf
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/bootloader_flash.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/bootloader_random.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/secure_boot_signatures.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/secure_boot.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/esp_image_format.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/flash_encrypt.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader_support/src/flash_partitions.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/log/log.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/spi_flash/spi_flash_rom_patch.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/micro-ecc/micro-ecc/uECC.o: No such file or directory
	xtensa-esp32-elf-gcc: error: build/bootloader//Users/pac/esp/xtensa-esp32-elf/bin/components/bootloader/src/main/bootloader_start.o: No such file or directory
	make: *** [build/bootloader.elf] Error 1

So it apears that the bootloader files for xtensa are not included ??

dpgeorge, 2017-02-27:
> ESPIDF=/Users/pac/esp/xtensa-esp32-elf/bin

No, that's wrong: ESPIDF needs to point to the *IDF*, not the compiler's path.

diginfo, 2017-02-27:
OOps, OK, yes that seems more logical given the variable name.

Now it fails after less than a second with:

undefined reference to `rtc_set_cpu_freq'

    ........
	LINK build/bootloader.elf
	build/bootloader//Users/pac/esp/esp-idf/components/bootloader/src/main/bootloader_start.o:(.literal.bootloader_main+0x74): undefined reference to `rtc_set_cpu_freq'
	build/bootloader//Users/pac/esp/esp-idf/components/bootloader/src/main/bootloader_start.o: In function `bootloader_main':
	bootloader_start.c:(.text.bootloader_main+0x6): undefined reference to `rtc_set_cpu_freq'
	bootloader_start.c:(.text.bootloader_main+0xf): undefined reference to `rtc_set_cpu_freq'
	collect2: error: ld returned 1 exit status
	make: *** [build/bootloader.elf] Error 1

dpgeorge, 2017-02-27:
> Now it fails after less than a second with:

You need to make sure you checkout the correct version of the ESP IDF:
```
$ git checkout 0a678ebe8c8bb78053f2a2dc4bd26124adde9122
```
Note that the build is giving you a warning about this, at the very start.

dpgeorge, 2017-02-27:
I just made an update to the code in 4df414b840daf75ed0e0d042848b786288149649 to support the latest ESP IDF version.

pacmac, 2017-02-27:
Thanks so much, but that's what I already did in my step by step above:

## Fix ESP-IDF Version
cd ~/esp/esp-idf
git checkout 0a678ebe8c8bb78053f2a2dc4bd26124adde9122

But I just ran it again with the following output:

    ~/esp/esp-idf
    $ git checkout 0a678ebe8c8bb78053f2a2dc4bd26124adde9122
    M	components/bt/lib
    M	components/esp32/lib
    HEAD is now at 0a678eb... Merge branch 'feature/openssl_option' into 'master'

And ran:

make clean
make

And I get the identical same error ?!

pacmac, 2017-02-27:
After fixing the path, I don't get the version error any more, here's the full output:

    pacBook:esp32 pac$ make clean
    Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
    rm -rf build 
    pacBook:esp32 pac$ make
    Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
    mkdir -p build/bootloader//Users/pac/esp/esp-idf/components/bootloader/src/main
    mkdir -p build/bootloader//Users/pac/esp/esp-idf/components/bootloader_support/src
    mkdir -p build/bootloader//Users/pac/esp/esp-idf/components/log
    mkdir -p build/bootloader//Users/pac/esp/esp-idf/components/micro-ecc/micro-ecc
    mkdir -p build/bootloader//Users/pac/esp/esp-idf/components/spi_flash
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/bootloader_random.c
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/secure_boot.c
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/esp_image_format.c
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c
    CC /Users/pac/esp/esp-idf/components/bootloader_support/src/flash_partitions.c
    CC /Users/pac/esp/esp-idf/components/log/log.c
    CC /Users/pac/esp/esp-idf/components/spi_flash/spi_flash_rom_patch.c
    CC /Users/pac/esp/esp-idf/components/micro-ecc/micro-ecc/uECC.c
    CC /Users/pac/esp/esp-idf/components/bootloader/src/main/bootloader_start.c
    LINK build/bootloader.elf
    build/bootloader//Users/pac/esp/esp-idf/components/bootloader/src/main/bootloader_start.o:(.literal.bootloader_main+0x74): undefined reference to `rtc_set_cpu_freq'
    build/bootloader//Users/pac/esp/esp-idf/components/bootloader/src/main/bootloader_start.o: In function `bootloader_main':
    bootloader_start.c:(.text.bootloader_main+0x6): undefined reference to `rtc_set_cpu_freq'
    bootloader_start.c:(.text.bootloader_main+0xf): undefined reference to `rtc_set_cpu_freq'
    collect2: error: ld returned 1 exit status
    make: *** [build/bootloader.elf] Error 1

dpgeorge, 2017-02-28:
Please try again from scratch: delete the micropython-esp32 repo and clone it again, then go to the ESP IDF directory and do: git checkout master; git submodule update --init.  Then try building uPy again.

pacmac, 2017-02-28:
Thanks so much for your help, I appreciate that.

I am now in a different location (office) so I decided to do a fresh install on a different machine and it works out of the box and I now have  a REPL prompt :-)

closed 2017-02-28 by diginfo