micropython/micropython-esp32 #31: GURU: LoadProhibited crash with socket when no bind address

pacmac, 2017-02-28:
A minor issue I know, but I thought I would report it:

This causes crash:

    import socket
    sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    sock.bind(('',80))

This does not:

    import socket
    sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    sock.bind(('0.0.0.0',80))

nickzoic, 2017-02-28:
OK, thanks for reporting it because that's how we find stuff!

This is missing error handling in my code: `""` isn't a valid address for `lwip_getaddrinfo` so it is returning an error code which modsocket is ignoring, leading to the usual C chaos.

Interestingly, Python2/3 on Linux anyway seems to special-case strings which might be IP addresses, eg: `""`, `"0"`, `"00000"`, `"0.0.0"` (but not `"0.0.0.0.0"`) ... so that's interesting and maybe we should be doing that as well :-).

nickzoic, 2017-08-01:
The "" case is covered by 1365f7f29 ... and no-one's complained about the others so I'm going to close this one off.

closed 2017-08-01 by nickzoic