micropython/micropython-esp32 #33: ADC Syntax

pacmac, 2017-02-28:
I can't seem to initiate the ADC, all of the following result in the same 'expecting a pin' error:

    import machine
    adc= machine.ADC('GP36')
    adc= machine.ADC('P36')
    adc= machine.ADC(4)
    adc= machine.ADC('GPI36')
    adc= machine.ADC('ADC1')
    adc= machine.ADC('ADC1_0')
    
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    ValueError: expecting a pin

dpgeorge, 2017-02-28:
Use: machine.ADC(machine.Pin('GP36')).

pacmac, 2017-02-28:
Thanks but that results in:

    machine.Pin('GP36')
    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    TypeError: can't convert str to int

machine.ADC(machine.Pin('GP36')) gives the same error ?!

dpgeorge, 2017-02-28:
Sorry, you'll need to use integers for pins, eg: machine.Pin(36).

pacmac, 2017-02-28:
Thanks, that works:

    import machine
    adc = machine.ADC(machine.Pin(36))
    print(adc.read())
    >> 4095

Another question on same topic, is it possible to internally read the VCC on the ESP32 as it is with the ESP8266 ??

nickzoic, 2017-02-28:
I can't see anything mentioned about this in the esp-idf docs ($ESP_IDF/docs/api/peripherals/adc.rst) ... although there's an internal Hall Effect sensor which is pretty weird (and not yet supported by MicroPython).

The ADC input pins have a range of 0-1 volt so you'd need to add a voltage divider between VCC (or V_BATT) and an ADC pin and sample it that way.

diginfo, 2017-02-28:
Thanks, I just checked the datasheet and could not see it mentioned, but I am sure I read it somewhere on a forum, pretty sure it was an arduino forum.

Interesting thing I just noticed in the datasheet is there is an internal temperature sensor, it would be nice to be able to access that.

nickzoic, 2017-02-28:
esp8266 had a VCC option on the ADC but I dunno, it wasn't all that useful unless you had Vcc == Vbatt.  The docs do say:

"ESP32 integrates two 12-bit SAR ("Successive Approximation Register") ADCs [...] The ADC driver API currently only supports ADC1 (9 channels, attached to GPIOs 32-39)."

So it is possible that the alleged ADC2 might have some such connection but there's no mention of it in the esp-idf docs at this point.

closed 2017-03-03 by pacmac

laurivosandi, 2017-05-12:
Hi, what about adding ADC2 support? Arduino libs seem to have implemented ADC2 support: https://www.esp32.com/viewtopic.php?t=955

dpgeorge, 2017-05-12:
> Hi, what about adding ADC2 support?

It's currently not recommended to expose ADC2 to the user because it's used for low-level control of the wifi and BT, see: https://github.com/espressif/esp-idf/issues/461

Avi-TelnT, 2017-12-29:
ESP32 has an internal Temperature sensor, How it can be read?

robert-hh, 2017-12-30:
There was a PR for that here: https://github.com/micropython/micropython-esp32/pull/192/files
with some discussions, whether it is useful or not.

idesignstuff, 2018-02-19:
ESP-IDF offers atten() function to change the voltage range measurable. 11dB makes 0 to 3.3V possible.
Is that going to be built into Micropython?

nickzoic, 2018-02-19:
I think it should already be there, something like:

adc = machine.ADC(machine.Pin(35))
adc.atten(machine.ADC_ATTN_11DB)

idesignstuff, 2018-02-19:
Awesome! Thank you.

Christian

pvanallen, 2018-03-01:
The correct attenuation is below. Note the period instead of underscore.

adc = machine.ADC(machine.Pin(35))
adc.atten(machine.ADC.ATTN_11DB)