micropython/micropython-esp32 #34: basic sensor support

diginfo, 2017-02-28:
Is there any forecast as to when basic sensor support such as dht, onewire and display (ssd1306) would be available ?

Also, I saw another thread related to this, but disable_irq() and enable_irq() is pretty much essential when doing bit-banging, right now none of my previous code that uses sensors seems to be working !

diginfo, 2017-03-02:
Hi, I am guessing that the developers are all busy with other projects, and I am trying to decide whether I should continue my testing and development on the esp32 given that there does not seem to be very much activity.

It looks like this is quite a long way from having the same capabilities as the existing esp8266 port, is there any timeframe or plan to get the port completed ??

Many Thanks

dpgeorge, 2017-03-03:
> I am trying to decide whether I should continue my testing and development on the esp32 given that there does not seem to be very much activity.

A lot of the activity is currently happenening elsewhere, eg at upstream micropython, and the ESP IDF, and these changes will filter down here soon enough.

For what you request, it's mostly just a matter of enabling the features.  I've just now pushed some commits to enable onewire, ds18x20 and dht modules.  I tested them (on ESP WROVER KIT) and they seem to work correctly.  The interface/API is exactly the same as the esp8266.

For ssd1306, it should work out-of-the-box.  Just copy the ssd1306.py file to the board and import it (found in drivers/display/ dir of this repo).

> It looks like this is quite a long way from having the same capabilities as the existing esp8266 port, is there any timeframe or plan to get the port completed ??

I don't think it'll ever be "completed" :)  The main outstanding differences to esp8266 is that ussl module doesn't fully work, and webrepl is not enabled because socket callbacks are difficult to implement.

pacmac, 2017-03-03:
Wow, brilliant, thank you so much, thats a very nice surprise that I was not expecting.

I have already verified that both dht and ds1820 are working well, I will also test ssd1306 in the next 2 days and feedback.

if only irq_enable(), irq_disable() and deepsleep() can also be added then I can convert all my existing 8266 code to work on the esp32.

Of course the big and the huge benefit on this device is dual core support and threading, when they are working this really will be a a very powerful, nice port.

Thank you again.

dpgeorge, 2017-03-03:
@pacmac good to hear it works for you.  Why do you need the irq enable/disable functions?  The esp32 is faster than the esp8266 so it might work without calling those functions.

pacmac, 2017-03-03:
I have a hx711 device (which is a load cell amplifier), it took quite a bit of tweaking to get it to run on the esp8266 using bit-banging, I have tried (although I admit not for a long time) to run the code on the esp32 by adjusting the main sleep_us() up and down but are unable to read it.

But I don't know exactly how much the interrupts are kicking in and so my conclusion was that it was the interrupts that were causing the problem.

I don't really want to have to re-write the driver as I would rather spend my time on developing the apps rather than the drivers, I will have another go in the next few days and see if I can crack it.

But without deepsleep() the 8266 can't really be replaced as in my current project it has to survive for at least 6 months on it's battery power, which on the 8266 is already achievable.

pacmac, 2017-03-03:
I just compiled the latest, nice bump on the heap size :-)

diginfo, 2017-03-08:
Thanks for adding the enable / disable_irq

closed 2017-03-08 by diginfo