micropython/micropython-esp32 #39: Using select to poll for a socket for incoming connections does not work

dpgeorge, 2017-03-03:
One should be able to use select.select([s], [], []) (or the equivalent select.poll call) to wait for an incoming connection on the socket.  But this doesn't work properly with esp32.

@nickzoic I saw that you implemented the socket_ioctl such that it used lwip_recvfrom to poll for read status, but uses select to poll for write/exception status.  What was the reason?  I think we should use select for read polling as well so to fix the issue with waiting for an incoming connection.

nickzoic, 2017-03-03:
Yeah, from memory that was because the simple and obvious 'select' way
didn't work ... from memory that bit never clears, so it always looks
like the socket is readable, but that's just from memory.I should have
included a comment there as a reminder,
I tell you what I'll pop it back to the obvious way and check it out ...

(there's also some chance that the IDF changes since then have fixed the
underlying problem, which would be even nicer since the MSG_PEEK thing
is a bit of a hack)

nickzoic, 2017-03-03:
OK, I just changed it back to do it the obvious way (without the MSG_PEEK workaround), and that seems to work fine now both as TCP client and TCP server ... and it works to check if a listening socket is ready to accept, so I think that solves your problem.  I'll submit it as a PR in a mo.

closed 2017-03-06 by dpgeorge