micropython/micropython-esp32 #41: After the last upgrade of esp-idf the compilation of micropython-esp32 fails

todork, 2017-03-03:
After the last upgrade of esp-idf to master - c62ae777c262aca1ad7b70c953e4c9ddde5df764 the compilation of the micropython-esp32 faild with:

> In file included from /home/vagrant/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h:30:0,
> from /home/vagrant/esp-idf/components/mbedtls/include/mbedtls/platform.h:29,
> from ../py/../extmod/modussl_mbedtls.c:39:
> ./sdkconfig.h:84:1: error: stray '##' in program
> ##define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 10
> ^
> ./sdkconfig.h:84:3: error: unknown type name 'define'
> ##define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 10
> ^
> ./sdkconfig.h:84:49: error: expected '=', ',', ';', 'asm' or 'attribute' before numeric constant
> ##define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 10
> ^
> ./sdkconfig.h:85:1: error: stray '##' in program
> ##define CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM 0
> ^
> ./sdkconfig.h:86:1: error: stray '##' in program
> ##define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM 32
> ^
> ./sdkconfig.h:87:1: error: stray '##' in program
> ##define CONFIG_NUMBER_OF_MAC_ADDRESS_GENERATED_FROM_EFUSE 4
> ^
> make: *** [build/py/../extmod/modussl_mbedtls.o] Error 1

To fix it I added in micropython-esp32/esp32/sdkconfig.h the following lines:

> #define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 10
> #define CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM 0
> #define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM 32
> #define CONFIG_NUMBER_OF_MAC_ADDRESS_GENERATED_FROM_EFUSE 4
> 

the compilation passed and seems that micropython is working.
Pls. check if somthing else is missing.

Regards

todork, 2017-03-04:
Sorry, I mentioned that I used not supported git hash.

dpgeorge, 2017-03-06:
> Sorry, I mentioned that I used not supported git hash.

I'm not sure what your issue is.  Which git hash of the ESP IDF did you use?  And which version of this repo?

nickzoic, 2017-03-06:
I think this is more of a patch to bring us up to supporting IDF master ... I can have a look and turn it into a PR if you like :-)

todork, 2017-03-06:
I am using usually the HEAD of esp-idf git (in this case 
c01dff82630d816c835fc7b4a7610c14684a25e8) when compiling micropython for 
esp32, but I hadn't problems up to the last update. After the 
compilation broke I saw that you support only given git hashes.

Regards,

Todor Kouyoumdjiev

ст. н.с. II ст.,
ст. чл. IEEE,
д-р Тодор Куюмджиев
Изп. Директор ЕКоКоМС ООД
тел.:    +359 2 9743929
факс.    +359 2 9743986
Уеб:     www.ecocoms.com
Скайп:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

Senior Research Fellow,
Senior Member IEEE,
Dr. Todor Kouyoumdjiev
CEO ECoCoMS Ltd.
Tel.:    +359 2 9743929
Fax.     +359 2 9743986
Web:     www.ecocoms.com
Skype:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

На 6.03.2017 в 07:04, Damien George написа:

todork, 2017-03-06:
I will appreciate this.

Regards,

Todor Kouyoumdjiev

ст. н.с. II ст.,
ст. чл. IEEE,
д-р Тодор Куюмджиев
Изп. Директор ЕКоКоМС ООД
тел.:    +359 2 9743929
факс.    +359 2 9743986
Уеб:     www.ecocoms.com
Скайп:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

Senior Research Fellow,
Senior Member IEEE,
Dr. Todor Kouyoumdjiev
CEO ECoCoMS Ltd.
Tel.:    +359 2 9743929
Fax.     +359 2 9743986
Web:     www.ecocoms.com
Skype:   tkouyoumdjiev
twitter: @tkouyoumdjiev
          @ECoCoMS

На 6.03.2017 в 10:19, Nick Moore написа:

dpgeorge, 2017-03-06:
> I think this is more of a patch to bring us up to supporting IDF master

I see, that makes more sense!

closed 2017-03-07 by dpgeorge