micropython/micropython-esp32 #44: can not compile base on the lastest mpy-esp32 and esp-idf

ouki-wang, 2017-03-07:
LINK build/application.elf
/home/ouki/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_mac_init+0x0): undefined reference to `esp_read_mac'
/home/ouki/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_mac_init+0xc): undefined reference to `esp_read_mac'
/home/ouki/esp/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o): In function `wifi_mac_init':
(.text.wifi_mac_init+0x17): undefined reference to `esp_read_mac'
make: *** [build/application.elf] Error 1

dpgeorge, 2017-03-07:
You probably need to do `git submodule update --init` in the ESP IDF directory.

aledwmorris, 2017-03-28:
+1 for `git submodule update --init`

I was getting thes linker errors relating to the RTC:

```
LINK build/application.elf
build//home/aledm/esp/esp-idf/components/esp32/cpu_freq.o:(.literal.esp_set_cpu_freq+0x10): undefined reference to `rtc_init_lite'
build//home/aledm/esp/esp-idf/components/esp32/cpu_freq.o: In function `esp_set_cpu_freq':
cpu_freq.c:(.text.esp_set_cpu_freq+0x12): undefined reference to `rtc_init_lite'
/home/aledm/esp/esp-idf/components/esp32/lib/librtc.a(pm.o):(.text+0x0): undefined reference to `rtc_set_wakeup_opt'
etc etc etc
```

and `git submodule update --init` made the errors go away.

nickzoic, 2017-04-04:
We're pretty certain this is the typical `submodule update --init` problem so I'm going to close this off.  @ouki-wang if you're still not able to build after following all the instructions through please let us know.

closed 2017-04-04 by nickzoic