micropython/micropython-esp32 #45: threading and power saving

diginfo, 2017-03-07:
I see that the pycom team have implemented power saving and threading.

Is it possible to roll these updates into this repo ?

diginfo, 2017-03-10:
Ping.....

Sorry to nag and for my impatience, but I would really like to convert some of my esp8266 battery powered projects to run on the ESP32.

Is power saving something that is likely to happen soon ??

dpgeorge, 2017-03-10:
Please understand that this port is still a work-in-progress and mostly driven by the community / volunteer work.  If you're really keen about deepsleep mode then please investigate how it works in the IDF and suggest a way for it to be implemented.

diginfo, 2017-03-10:
Thanks Damien, I appreciate and understand that, but I am afraid that that is beyond my capabilities / skills.

But it's OK, never mind, I will close all these open issues / requests and forget about the ESP32 device for the foreseeable future.

Thanks again

closed 2017-03-10 by diginfo