micropython/micropython-esp32 #46: Compile Latest Fails

diginfo, 2017-03-07:
I just tried compiling the latest and it fails as follows:

	......
	.....
	.....
	CC machine_dac.c
	CC modmachine.c
	CC modnetwork.c
	In file included from modnetwork.c:43:0:
	modnetwork.c: In function 'esp_initialize':
	/Users/pac/esp/esp-idf/components/esp32/include/esp_wifi.h:104:19: error: 'CONFIG_ESP32_WIFI_RX_BUFFER_NUM' undeclared (first use in this function)
	     .rx_buf_num = CONFIG_ESP32_WIFI_RX_BUFFER_NUM, \
	                   ^
	modnetwork.c:196:34: note: in expansion of macro 'WIFI_INIT_CONFIG_DEFAULT'
	         wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	                                  ^
	/Users/pac/esp/esp-idf/components/esp32/include/esp_wifi.h:104:19: note: each undeclared identifier is reported only once for each function it appears in
	     .rx_buf_num = CONFIG_ESP32_WIFI_RX_BUFFER_NUM, \
	                   ^
	modnetwork.c:196:34: note: in expansion of macro 'WIFI_INIT_CONFIG_DEFAULT'
	         wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	                                  ^
	make: *** [build/modnetwork.o] Error 1

dpgeorge, 2017-03-08:
When you downgrade the ESP IDF you need to do `git submodule update --init`.

diginfo, 2017-03-08:
OOps, yes that works, I have added that step to my procedure and won't make the same mistake again.

Thanks so much

closed 2017-03-08 by diginfo

zmijunkie, 2017-09-09:
Happy to have seen this issue - and it's solution. REPL Fun started a few minutes later ;-)