micropython/micropython-esp32 #47: esp32: tcp data transfer fails with new build

robert-hh, 2017-03-09:
I have a little test script excerpted from a large script, that worked until build 3d903f5b3eb9305, and now fails with 110, ETIMEDOUT at the line, where it should send. Do I have to change my script or is this an error. The script works on esp8266 and LoPy.
```
import socket
DATA_PORT = 13333
datasocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
datasocket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
datasocket.bind(socket.getaddrinfo("0.0.0.0", DATA_PORT)[0][4])

datasocket.settimeout(None)
datasocket.listen(1)

dataclient, data_addr = datasocket.accept()
print("FTP Data connection from:", data_addr)
try:
    dataclient.send("This is a little bit of a message")
except Exception as err:
    print(err)

dataclient.close()
datasocket.close()
```

nickzoic, 2017-03-10:
G'day Robert,

I'm at the beach typing with thumbs right now so can't try it out, but can you try adding

`dataclient.settimeout(None)`

just after the `accept` ... the child socket doesn't inherit the timeout from the listening one ...

dpgeorge, 2017-03-10:
The bug was that accepted sockets did not have their timeout initialised upon creation, and the retries variable had garbage value in it.  Fixed in 741918ae51dfe91940be45445c5902d0fbbba87a by initialising the value.

closed 2017-03-10 by dpgeorge