micropython/micropython-esp32 #48: esp32: TCP data transfer to board does not complete

robert-hh, 2017-03-10:
Hello @dpgeorge and @nickzoic, thanks for fixing the last issue. If I go one step further, I run again into glitch. Extending my little test script as below, the transfer does not finish. The last partial block does not arrive. Build 3d903f5b3eb930594a and ESP8266 work. On the PC, I run:
`cat testfile| nc -q 1 ip_address 13333`

```
import socket

DATA_PORT = 13333

datasocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
datasocket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
datasocket.bind(socket.getaddrinfo("0.0.0.0", DATA_PORT)[0][4])

datasocket.settimeout(None)
datasocket.listen(1)

dataclient, data_addr = datasocket.accept()
print("FTP Data connection from:", data_addr)

with open("testfile", "w") as file:
    chunk = dataclient.read(512)
    while len(chunk) > 0:
        file.write(chunk)
        chunk = dataclient.read(512)

dataclient.close()
datasocket.close()
```

robert-hh, 2017-03-13:
Hello @dpgeorge and @nickzoic,
the bug can be fixed by changing line 368 of modsocket.c from
`        if (x = 0) return x;`
to
`        if (x >= 0) return x;`

dpgeorge, 2017-03-14:
Thanks for the fix!  Patched in 8ad6df61ef4830bcb172d43bb89220f4b913abc8

closed 2017-03-14 by dpgeorge