micropython/micropython-esp32 #49: RTCWDT_RTC_RESET endless loop

pacmac, 2017-03-15:
I just received some ESP32-WROOM generic development boards and after make, make erase and make deploy and restarting the device it is stuck in an endless loop:

	rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
	configsip: 0, SPIWP:0x00
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:QIO, clock div:2
	load:0x3fff0008,len:8
	load:0xffffffff,len:-1
	ets Jun  8 2016 00:22:57
	
Any suggestions on what the cause may be ??

dpgeorge, 2017-03-15:
You might need to use `FLASH_MODE = dio` and `FLASH_SIZE = 2MB` config settings in your GNUmakefile.

pacmac, 2017-03-15:
Thanks so much, yes indeed that solved it, I now have the REPL :-)

closed 2017-03-15 by pacmac

rustyx, 2018-11-29:
Also make sure watchdog is disabled if your image doesn't support it.

```
CONFIG_BOOTLOADER_WDT_ENABLE=n
```