micropython/micropython-esp32 #52: Bluetooth not implemented

mahanet, 2017-03-20:
I would like to be able to scan, connect and send/receive data via Bluetooth.
Anyone started working on that?

MrSurly, 2017-03-24:
I'm going to take a stab; I need it as well.

@dpgeorge Any hints on where a BT module should live in the namespace?  `machine`, or `esp`?  Somewhere else?

mahanet, 2017-03-24:
At Pycom they put it under network-
from network import Bluetooth

MrSurly, 2017-03-24:
That seems reasonable.

MrSurly, 2017-03-24:
> At Pycom they put it under network-from network import Bluetooth

Hold on; if pycom has already implemented this, why not roll that back into micropython-esp32?

Could @nickzoic  or @dpgeorge comment?

pfalcon, 2017-03-25:
@MrSurly : Learn about open-source licensing?

dpgeorge, 2017-03-25:
> Hold on; if pycom has already implemented this, why not roll that back into micropython-esp32?

I believe that Pycom are still intending to contribute to this port (using an MIT license) and bluetooth may be part of that contribution, but I can't say for sure and I don't know what their timeline is for contributing (I expect it to be soon though).

MrSurly, 2017-03-26:
@dpgeorge Looks like Pycom has retained the MIT license (as well as adding a slightly modified GPLv3 license).  Are you saying there is a licensing issue, or that it would be preferable for Pycom to to do the contribution, from a technical standpoint?

dpgeorge, 2017-03-27:
> Looks like Pycom has retained the MIT license

Not quite sure what you mean... if I look at their BT file https://github.com/pycom/pycom-micropython/blob/master/esp32/mods/modbt.c it's only GPLv3 licensed.  So that's a licensing issue meaning we'd need for Pycom to contribute the code.

MrSurly, 2017-03-27:
I was referring to the license file in the root of the repository. I guess
the source file has precedence.

dpgeorge, 2017-03-27:
> I was referring to the license file in the root of the repository.

I see... well that's the license that I added long ago!  They really should change that file to make it explicit that the MIT license does not cover their new files.

pfalcon, 2017-03-27:
> I see... well that's the license that I added long ago! They really should change that file to make it explicit that the MIT license does not cover their new files.

That's not how viral (vigorously freedom-protecting) licenses work. The mere presence of GPL notice anywhere in the project makes the entire project GPL, regardless of what else may be written somewhere else. Somebody (party A) who doesn't know that 100%, should not do anything about that code (even look into it) without consulting the licensing lawyer. Then, depending on the outcome of the consultation, the party A should be prepared to provide such a consultation (of an official licensing lawyer) to any other party which may be affected by their work (entire world in the case of open-source software), because the incorrect initial consultation and actions of party A stemming from it (contributing wrong code to wrong project), may put thousands and thousands of unsuspecting parties under a liability.

P.S. @dpgeorge: The over-careless license (like, not Apache2.0) of MicroPython and lack of explicit sign-off for each change may become a problem for adoption of the project (well, maybe it became already).

MrSurly, 2017-03-27:
> I see... well that's the license that I added long ago! They really should change that file to make it explicit that the MIT license does not cover their new files.

I agree.  I'm new to μPy (though not to microcontrollers or Python).  I don't really have any intention or desire to stir up licensing issues or to disrupt any goodwill between μPy and Pycom. 

As such, I'm really hesitant to reach out to Pycom WRT clarification, since the μPy community is fairly small, and (I assume) somewhat tight-knit.  Treading lightly is preferred.

I just want to get my ESP32 boards up-and-running to their full potential (which includes Bluetooth).  As such, it comes to an old dilemma: Wait for a downstream implementer to "give back" to the upstream project, or to just begin re-inventing the wheel.  The latter choice is really _not_ at all preferable. 

Do you have any advice on moving forward with this that wouldn't cause ... problems?

dpgeorge, 2017-03-31:
> The over-careless license (like, not Apache2.0) of MicroPython and lack of explicit sign-off for each change may become a problem for adoption of the project

@pfalcon if you have concerns about this please raise an issue at the main repo so we can discuss it and work out a solution, thanks.

pfalcon, 2017-03-31:
@dpgeorge : Thanks for the reply. Well, my concern probably doesn't go as far as to raise a separate issue on that - not before I actually hear such concerns from other folks vs it being just my speculation. Nor I believe we can do much about it. However, I'm sure we should perform all the due diligence in regard to the current license, this includes being prompt on warning contributors that GPL code should not come anywhere close to the mainline (or official forks). That's why I, while appreciating @MrSurly's growing contribution efforts, considered it important to issue such a warning.

dpgeorge, 2017-04-01:
> However, I'm sure we should perform all the due diligence in regard to the current license

Yes I fully agree.  I always have this in the back of my mind and I think for the most part the contributions we have are small patches to existing code (that probably you or I wrote from scratch) and these shouldn't be a problem.  What's more to worry about is when new files are committed, or large slabs of code, and there we just need to start being more careful, I guess, by educating contributors.

MrSurly, 2017-04-01:
Well, for my part, I'm not looking at Bluetooth, just moving on to other
areas (I.e. deep sleep and SPI). If I do tackle BT, I'll work it out from
scratch, or the IDF examples.

Actually, the new files I'm working on are ones that I copied the structure
from other parts of the project. I guess I should update the top part with
my own name. 😀

pfalcon, 2017-04-03:
@dpgeorge , Thanks.

@MrSurly:

> If I do tackle BT, I'll work it out from scratch, or the IDF examples.

Thanks.

>  I guess I should update the top part with my own name. :grinning:

Yes, if you wrote a file from scratch, or made a significant contribution to one, definitely add your copyright.

MrSurly, 2017-04-03:
Starting to look at Bluetooth, now

MrSurly, 2017-04-05:
Okay, looking at what changes are made to `sdkconfig` when using `make menuconfig` on an empty ESP-IDF project.

These were the only changes:

```
CONFIG_BT_ENABLED=y
CONFIG_BLUEDROID_ENABLED=y
CONFIG_BTC_TASK_STACK_SIZE=3072
CONFIG_BT_RESERVE_DRAM=0x10000
```

I've duplicated this in `esp32/sdkconfig.h`, and I've reduced the μPy heap size to 64K (and then to 32K), in the hopes that it would prevent the WDT from firing, as mentioned by @dpgeorge [here](https://github.com/espressif/esp-idf/issues/487#issuecomment-291712054)

Reducing the heap prevented the 300ms WDT from firing and `esp_bt_controller_init()` does return now. But the 5s WDT is still triggering:

```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2372
load:0x40078000,len:6708
load:0x40080000,len:252
entry 0x40080034
I (1526) cpu_start: Pro cpu up.
I (1526) cpu_start: Single core mode
I (1527) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1540) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1561) heap_alloc_caps: At 3FFD1678 len 0000E988 (58 KiB): DRAM
I (1581) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1603) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1624) heap_alloc_caps: At 4009A898 len 00005768 (21 KiB): IRAM
I (1645) cpu_start: Pro cpu start user code
I (1706) cpu_start: Starting scheduler on PRO CPU.
```

I know that μPy has some custom LD scripts, wondering if anyone here can comment on whether the `CONFIG_BT_RESERVE_DRAM` setting (which is used by esp32.ld from the IDF) impinges upon other RAM allocations in the custom scripts?

dpgeorge, 2017-04-06:
@MrSurly I took a look at your issue and found the problem, to do with ISR numbers (both BT and UART use ISR number 5).  I fixed the issue in a3fd435d0f0efeec28f003f5d5c002dbee51073b.  You should now be able to start BT successfully.

MrSurly, 2017-04-06:
That explains why the REPL console locked for the 5 seconds before the WDT kicks in.

How on earth did you find that? 

Here, I was thinking that BT had to be pinned to a particular core, as seen in the `blue_adv` example from the IDF.

dpgeorge, 2017-04-06:
> How on earth did you find that?

If the code didn't return to the REPL then it worked fine, the BT actually worked.  Only after pressing a key (ie the UART RX being activated) was there a problem.

MrSurly, 2017-04-06:
@dpgeorge a3fd435d0f0efeec28f003f5d5c002dbee51073b works for me, thanks!

MrSurly, 2017-04-11:
Okay, so I'm still moving (slowly) forward on this.

## The short version: 

I had early success in initializing the BT subsystem, but the _full_ BT stack makes the code about 40% larger, and it doesn't run at all.

## The long version:

##### Preface: How I know it might work, and it's not just a `sdkconfig.h` error:

If I only do these two calls (with the minimum necessary object files to make it work):
```c
        esp_bt_controller_init(); 
        switch(esp_bt_controller_enable(ESP_BT_MODE_BTDM))
```
The BT subsystem actually comes up (you can see the log messages, similar to the wifi). Image size is just a little larger than without BT.

##### What I suspect

That the flash image, in conjunction with the FAT configuration, and possibly the heap, are simply too large.  I can't find a definitive answer on the WROOM-32 flash size, nor can I figure out how to resize the FAT FS.

##### The rest of the 'long version'

I've added the 110 (!) object files necessary to have compile with some very basic BLE functionality.  Essentially, just enable BT, and do some advertising.

Doing this makes the binary grow to about 130000 to 1440000 bytes; the latter being when I actually call to any BT functions _at all_.

I either have two different symptoms.  For smaller images:
```
FAT filesystem appears to be corrupted. If you had important data there, you
may want to make a flash snapshot to try to recover it. Otherwise, perform
factory reprogramming of MicroPython firmware (completely erase flash, followed
by firmware programming).
```

So, I figure I'll reduce the FAT file system size, though I can't seem to find where this is configured.

For larger binaries (where I'm actually calling `esp_bt...` or `esp_ble...` functions), I get 

```ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2372
load:0x40078000,len:6708
load:0x40080000,len:252
entry 0x40080034
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2372
load:0x40078000,len:6708
load:0x40080000,len:252
entry 0x40080034
I (1858) cpu_start: Pro cpu up.
I (1858) cpu_start: Single core mode
I (1859) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1872) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1892) heap_alloc_caps: At 3FFDD630 len 000029D0 (10 KiB): DRAM
I (1913) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1935) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1956) heap_alloc_caps: At 4009B1FC len 00004E04 (19 KiB): IRAM
I (1977) cpu_start: Pro cpu start user code
I (2038) cpu_start: Starting scheduler on PRO CPU.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400800f9  PS      : 0x00060530  A0      : 0x80007c34  A1      : 0x3ffe3e70  
A2      : 0x3ff0005c  A3      : 0x40006840  A4      : 0x3ff10000  A5      : 0x000008ff  
A6      : 0x000008ef  A7      : 0x000008ef  A8      : 0x800800f9  A9      : 0x3ffe3d10  
A10     : 0x000008e6  A11     : 0x3ff0005c  A12     : 0x000008ef  A13     : 0x3f400000  
A14     : 0x00020000  A15     : 0x0000000e  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400800f9:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20
```

I've tried `make clean` and `make erase`.   Here's a summary:

* No BT object files (or BT calls), binary around 1M: *WORKS*
* Basic BT object files (just turn on BT), binary around 1M: *WORKS*
* All  BT object files, no BT calls, binary around 1.3M, heapsize 48K: FAT errors
* All BT object files, no BT calls, binary around 1.3M, heapsize 64K: continuous reboots
* All BT object files, BT calls, binary around 1.44M, any heapsize: continuous reboots

The whole mess is at https://github.com/MrSurly/micropython-esp32/tree/39b25b3f5da3d4f6edb0f3615ddf27b62ffeafc8

dpgeorge, 2017-04-12:
Ok, if the BT stack requires that much code space then we need to increase the amount available for the firmware, and adjust the filesystem.

Right now there is a maximum of 1MiB for the firmware, which is why you see problems when the firmware grows above that.  And the filesystem is 256k.

Since the reference HW (ESP-WROOM-32) has 4MiB of flash, and I doubt anyone would make a module with less flash than that, I suggest to increase the available flash for the firmware to 2MiB (which should be plenty for BT and more) and increase the filesystem to 2MiB.

Atzingen, 2017-07-16:
So what's the status on the BLE on esp32 ?

MrSurly, 2017-07-16:
> So what's the status on the BLE on esp32 ?

It's being discussed in [PR #86](https://github.com/micropython/micropython-esp32/pull/86), with likely API changes.  Comments should be made there.

Atzingen, 2017-07-16:
Sorry, All ask there. Thanks !

dpgeorge, 2019-09-25:
BLE support for esp32 is being implemented upstream via the following PR: micropython/micropython#5051

closed 2019-09-25 by dpgeorge