micropython/micropython-esp32 #58: Implement HW SPI

MrSurly, 2017-03-30:
Created to file a PR against.

I have this [working](https://github.com/MrSurly/micropython-esp32/tree/dev-hwspi). Need to test a bit more, and go over the code to make sure it's worthy.

Also, would like to get feedback on [micropython/micropython #2986](https://github.com/micropython/micropython/issues/2986) before filing a PR

dpgeorge, 2017-03-30:
Re the hspi driver: please call it something that mirrors the MCU.  Do Espressif call it HSPI?  Maybe hard_spi is a better name (see stmhal/spi.c at the bottom of that file for more hints).

MrSurly, 2017-03-31:
> Re the hspi driver: please call it something that mirrors the MCU. Do Espressif call it HSPI? Maybe hard_spi is a better name (see stmhal/spi.c at the bottom of that file for more hints).

I'm assuming this only applies to the `.c` and `.h` files, right?  Presently, I'm calling them `machine_hspi.c` and `machine_hspi.h`.  I personally don't care at all what they're called, but since there seemed to be some precedence for the "hspi" name, I kept that.  Coincidentally (see below), one of the SPI controllers on the ESP32 is actually called `HSPI`.

My personal inclination would be to name them `machine_hardware_spi.*` or `machine_hw_spi.*`.

The ESP32 has two SPI  "hosts" (as they're called) that can be accessed via the IDF: HSPI and VSPI.  Interestingly, you can use any pins (or none!) that you like for your SPI host -- the ESP32 has an internal routing matrix (two, actually ...)

Is it safe to assume that a PR for this would be welcome in the near future?

dpgeorge, 2017-03-31:
> I'm assuming this only applies to the .c and .h files, right?

To everything, the file names and the C identifiers (function names etc) in those files.

> The ESP32 has two SPI "hosts" (as they're called) that can be accessed via the IDF: HSPI and VSPI.

In that case the uPy bindings should *not* be called hspi because it would be confusing.  The uPy bindings apply to both the HSPI and VSPI (I assume).  Calling it machine_hw_spi sounds good.

> Is it safe to assume that a PR for this would be welcome in the near future?

Yes!

MrSurly, 2017-03-31:
Okay, the ESP-IDF implementation of HW SPI appears to have some issues: [SPI: pin routing remains after disabling SPI](https://github.com/espressif/esp-idf/issues/478), and [spi_device_transmit doesn't return when changing host ID](https://github.com/espressif/esp-idf/issues/477).

I'm hoping the ESP IDF team is responsive.

These two issues are probably blockers; HW SPI works fine as long as you configure it exactly _once_, which is probably OK for many applications.

MrSurly, 2017-04-03:
PR #63

closed 2017-04-06 by MrSurly