micropython/micropython-esp32 #59: Whether the current version have file system function support

YanMinge, 2017-03-31:
when setup the system, I can get the log:

```
could not open file 'boot.py' for reading
could not open file 'main.py' for reading
```
and I try to add a function mptask_init_sflash_filesystem like pycom-micropython
```
STATIC void mptask_init_sflash_filesystem (void) {
    FILINFO fno;

    // Initialise the local flash filesystem.
    // Create it if needed, and mount in on /flash.
    FRESULT res = f_mount(&sflash_fatfs);
    if (res == FR_NO_FILESYSTEM) {
        // no filesystem, so create a fresh one
        res = f_mkfs(&sflash_fatfs, FM_SFD | FM_FAT, 0, NULL, 0);
        if (res != FR_OK) {
            printf("no file sys,failed to create flash\n");
        }
        // create empty main.py
        mptask_create_main_py();
    }
    else if (res == FR_OK) {
        // mount sucessful
        if (FR_OK != f_stat(&sflash_fatfs,"/flash/main.py", &fno)) {
            // create empty main.py
            mptask_create_main_py();
        }
    } else {
        printf("failed to create flash,res:%d\n",res);
    }

    // create /flash/sys, /flash/lib and /flash/cert if they don't exist
    if (FR_OK != f_chdir (&sflash_fatfs,"/flash/sys")) {
        f_mkdir(&sflash_fatfs,"/flash/sys");
    }
    if (FR_OK != f_chdir (&sflash_fatfs,"/flash/lib")) {
        f_mkdir(&sflash_fatfs,"/flash/lib");
    }
    if (FR_OK != f_chdir (&sflash_fatfs,"/flash/cert")) {
        f_mkdir(&sflash_fatfs,"flash/cert");
    }

    f_chdir (&sflash_fatfs,"/flash");

    // make sure we have a /flash/boot.py. Create it if needed.
    res = f_stat(&sflash_fatfs,"/flash/boot.py", &fno);
    if (res == FR_OK) {
        if (fno.fattrib & AM_DIR) {
            // exists as a directory
            // TODO handle this case
            // see http://elm-chan.org/fsw/ff/img/app2.c for a "rm -rf" implementation
        } else {
            // exists as a file, good!
        }
    } else {
        // doesn't exist, create fresh file
        FIL fp;
        f_open(&sflash_fatfs,&fp, "/flash/boot.py", FA_WRITE | FA_CREATE_ALWAYS);
        UINT n;
        f_write(&fp, fresh_boot_py, sizeof(fresh_boot_py) - 1 /* don't count null terminator */, &n);
        // TODO check we could write n bytes
        f_close(&fp);
    }
}
```
f_mount will return error:  FR_NOT_READY,           /* (3) The physical drive cannot work */
I would like to ask how to operate the file additions, deletions, modify..?

robert-hh, 2017-03-31:
As far as I could tell, the current build has a file system in flash. Not the largest one indeed. boot.py and main.py may not be present, but you can create them. And an empty boot.py should be created by default.  Maybe you should try to redo erasing and creating & flashing the firmware.
If you are talkding about SD card support, look at this question: https://github.com/micropython/micropython-esp32/issues/55

YanMinge, 2017-03-31:
Thank you for your reply, I do not want to support sd card, just want to use C language to create a main.py file, or modify the boot.py.   and the function mount will return error.

robert-hh, 2017-03-31:
Is there a specific reason to do that in C and not in Python. if you look into esp32/modules, there is _boot.py which is executed first, before boot.py and main.py. And that calls setup() of inisetup.py, if no file system is found. So any boot time modification can be done there. And _boot.py is executed from flash.

YanMinge, 2017-04-01:
Yes，I know _boot.py can do this work，but I wan't use ftp or serial to update the Python file

dpgeorge, 2017-04-01:
> just want to use C language to create a main.py file, or modify the boot.py

uPy now includes a VFS subsystem and so you're not supposed to access the fatfs functions directly because the underlying filesystem might be something else (eg SPIFFS, ext4, etc).  This is actually a great feature because it means any code you write (eg an FTP server) will work with any filesystem type, not just FAT.

It's best to write code in Python, but if you really need to use C then you have 2 options:
- Use the mp_vfs_xxx functions (see extmod/vfs.h).  Note that they will raise an exception if they fail.  To write to files you need to use the stream interface (mp_stream_write...).
- Use the fatfs functions directly, but this only works if the filesystem is FAT.  See cc3200/mptask.c and cc3200/ftp/ftp.c for examples of how to do this.

YanMinge, 2017-04-01:
Thanks for you help, This information is helpful to me, I will try your suggestion.

closed 2017-04-01 by YanMinge