micropython/micropython-esp32 #65: Proposed code styling via Astyle

MrSurly, 2017-04-04:
For a PR #64 , I used `astyle -KSUHp -s4`.  Propose adding a single sentence to `CODECONVENTIONS.md`, recommending it's usage.

dpgeorge, 2017-04-05:
I've never used astyle.  Is it popular / well known?

MrSurly, 2017-04-05:
Well known among the developers I know =)  Available for Linux, Mac, Windows.  In most Linux repos, and installable on Mac easily via Brew.  Just thought this is one of those thing's that's easy to automate, or make easy for others.

There's also [indent](https://www.gnu.org/software/indent/).

nickzoic, 2017-08-01:
No objection to using astyle as we go along, it likes whitespace more than I do but at least it leaves most everything else alone.

dpgeorge, 2019-01-16:
Please see discussion upstream: https://github.com/micropython/micropython/issues/4223

closed 2019-01-16 by dpgeorge