micropython/micropython-esp32 #66: _thread module dev-status

lapsule, 2017-04-06:
Is there any plan for _thread module from PyCom?

lapsule, 2017-04-07:
according to docs, after adding mpthreadport.c/h files, got a crush, here is coredump info, looks like stack management issue: @dpgeorge 

```shell

===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x400dc702	0x400dc702 <mp_stack_set_top+6>
lbeg           0x400014fd	1073747197
lend           0x4000150d	1073747213
lcount         0xfffffffa	4294967290
sar            0x14	20
ps             0x60f20	397088
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x400ee4c5	1074717893
a1             0x3ffd5f30	1073569584
a2             0x3ffd5f50	1073569616
a3             0x0	0
a4             0x60823	395299
a5             0x0	0
a6             0x3ffca7e4	1073522660
a7             0x0	0
a8             0x800dc702	-2146580734
a9             0x3ffd5f10	1073569552
a10            0x0	0
a11            0x40080cc0	1074269376
a12            0x0	0
a13            0x40e	1038
a14            0x3ffd5f20	1073569568
a15            0x3ffb0a34	1073416756

==================== CURRENT THREAD STACK =====================
#0  0x400dc702 in mp_stack_set_top ()
#1  0x400ee4c5 in mp_task ()

======================== THREADS INFO =========================
  Id   Target Id         Frame
  5    process 4         0x40084b92 in xQueueGenericReceive ()
  4    process 3         0x40085604 in vTaskDelete ()
  3    process 2         0x40086170 in prvTimerTask ()
  2    process 1         0x400857ac in prvIdleTask ()
* 1    <main task>       0x400dc702 in mp_stack_set_top ()

======================= ALL MEMORY REGIONS ========================
Name   Address   Size   Attrs
.rtc.text 0x400c0000 0x0 RW
.iram0.vectors 0x40080000 0x400 R XA
.iram0.text 0x40080400 0x1bce8 R XA
.dram0.data 0x3ffb0000 0x1bc4 RW A
.flash.rodata 0x3f400010 0x2d4a8 RW A
.flash.text 0x400d0018 0x8e03c R XA
.coredump.tasks 0x3ffd5ff8 0x174 RW
.coredump.tasks 0x3ffd5e70 0x17c RW
.coredump.tasks 0x3ffd13f0 0x174 RW
.coredump.tasks 0x3ffd12c0 0x124 RW
.coredump.tasks 0x3ffd1bac 0x174 RW
.coredump.tasks 0x3ffd1af0 0xb0 RW
.coredump.tasks 0x3ffafce8 0x174 RW
.coredump.tasks 0x3ffafc10 0xcc RW
.coredump.tasks 0x3ffaeb64 0x174 RW
.coredump.tasks 0x3ffaea70 0xe8 RW

===================== ESP32 CORE DUMP END =====================
===============================================================

```

dpgeorge, 2017-04-08:
Pycom are intending to merged their code so, so I'd wait for that.

dpgeorge, 2017-04-13:
Threading is now supported on the esp32, via the _thread module.

closed 2017-04-13 by dpgeorge