micropython/micropython-esp32 #72: #59

YanMinge, 2017-04-18:
I have previously submitted an issue [#59](https://github.com/micropython/micropython-esp32/issues/59),
and I have done some work on vfs_fat and oofatfs, now I can use C language to create, modify, or delete files. but still some information I want to confirm with you.

1. I tried to modify the boot.py in micropython-esp32\esp32\modules\inisetup.py. But it looks like it can not work.
```
def setup():
    check_bootsec()
    print("Performing initial setup")
    uos.VfsFat.mkfs(bdev)
    vfs = uos.VfsFat(bdev)
    uos.mount(vfs, '/flash')
    uos.chdir('/flash')
    with open("boot.py", "w") as f:
        f.write("import Undefined")
    with open("main.py", "w") as f2:
        f2.write("import Undefined2")
    return vfs
```
also when I use  f_open and f_read from oofatfs/ff.c to read an existing boot.py.  I should set it as this
```
      f_chdir (&mb_sflash_vfs_fat->fatfs,"/flash");
      f_open(&mb_sflash_vfs_fat->fatfs,&fp2, "boot.py", FA_WRITE |FA_READ); 
      f_read(&fp2, mb_read_buffer, 62 /* don't count null terminator */, &n2);
      f_close(&fp2);
```
if I set as this, I can't read any thing.
```
      f_chdir (&mb_sflash_vfs_fat->fatfs,"/flash");
      f_open(&mb_sflash_vfs_fat->fatfs,&fp2, "boot.py", FA_WRITE |FA_READ|FA_CREATE_ALWAYS); 
      f_read(&fp2, mb_read_buffer, 62 /* don't count null terminator */, &n2);
      f_close(&fp2);
```
Have you tested the implementation of boot.py or main.py in python?  and why fs_open in oofatfs use unusual？

dpgeorge, 2017-04-19:
I'm not quite sure what you are asking... but the code you have should work just fine.  main.py and boot.py have been tested and work.

The exception that you get, "ImportError: no module named 'Undefined'" is expected because there is no module called "Undefined", and that's what boot.py is trying to import.

dpgeorge, 2017-04-19:
To check if boot.py is created just do:
```python
>>> f=open('boot.py')
>>> print(f.read())
>>> f.close()
```

YanMinge, 2017-04-19:
Yes, as you said, I change the boot.py. so if boot.py is executed, I expect to have the log
```
Traceback (most recent call last):
  File "_boot.py", line 3, in <module>
ImportError: no module named 'Undefined'
```
But I didn‘t get any error log like this.

dpgeorge, 2017-04-19:
> I change the boot.py with code in inisetup.py,

inisetup.py is only run if there is no existing filesystem on your board.  To make it execute you'll need to do "make erase" before doing "make deploy".

YanMinge, 2017-04-19:
Got it, I ignored the contents of _boot, sorry to waste your time.

closed 2017-04-19 by dpgeorge