micropython/micropython-esp32 #74: Low rate of UDP package

filipeaguiar22, 2017-04-22:
Hi everyone,

 I am trying to short the latency to the minimum using WiFi. 
 Using a very simple code, I found that the max rate of UDP packages I can send in one per 25ms. If I lower the delay, the I get the error OSError:ENOMEM (out of memory).
 What do you think can be done to increase the frequency?

Filipe
`

import network
import time

nic = network.WLAN(network.STA_IF)
nic.active(True)
nic = network.WLAN(network.STA_IF) # create station interface
nic.connect('MOTOROLA-46320', 'mariamaria') # connect to an AP
time.sleep(5)

#send udp:
import socket
s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
#s = socket.socket(socket.AF_INET, socket.SOCK_STREAM) 
#s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
address = ('192.168.10.8', 57029)
#address = ('192.168.10.8', 57855)
#s.setblocking(False)
s.connect(address) 

print(s)
print("Bombarding started")
k=0
while (True):
data = b'hello udp'
k=k+1
data = str.encode("i: {}".format(k))
s.sendall(data)
time.sleep(0.025)`

MrSurly, 2017-04-27:
This is just an off-the-cuff answer, but what happens if you move `data = b'hello udp'` above the loop, and eliminate the `data = str.encode("i: {}".format(k))` line altogether? 

Does the out-of-memory problem go away?

kfricke, 2017-04-27:
You should consider to only send the raw value of `k` in the UDP packet. To strictly type-cast it you can use the comfortable [uctypes module](http://docs.micropython.org/en/latest/esp8266/library/uctypes.html) (sry for the ESP8266 link, but ESP32 has no official docs yet).

The OOM exception is probably triggered by the altering speed of the loop before the Micropython garbage collector is called to do it's work. One can consider to call it manually (by calling `gc.collect()`) or avoid memory allocation in such critical areas.

A final note on your code example: As the ESP32 beeing the successor of the ESP8266 platform... It is actively promoted to close filehandles/ressources/sockets as soon as possible after used and needed on the ESP8266 platform. We should assume this here as well until told differently.
Your code does not contain the handling of the socket to be closed. Missing this may yield later problems!

_Please use MicroPython for everything but the next IoT DDoS botnet!_

closed 2019-01-16 by dpgeorge