micropython/micropython-esp32 #75: PWM support

vandys, 2017-05-05:
ESP8266 has a PWM module in machine; is there a reason it's missing on ESP32 (aside from
just time needed to get to it!).  I'm looking at pulling over the ESP8266 code and adapting,
unless there's a reason it won't work?

nickzoic, 2017-05-05:
I'm actually quite surprised by this because I thought I had already
implemented it!
Go for it ... probably the easiest way is to look at machine_adc.c and
use that same approach, and end up with the same API as ESP8266.  Let me
know if you get stuck on anything!

vandys, 2017-05-05:
Hi, thanks.  On it.  The API makes it look like you can specify PWM frequency on each PIN, but
the hardware just has a common frequency (well, one high speed and one low).  Do I just let
the frequency get overwritten each time?  I'm thinking about just using one timer, maybe the
low speed one.  Duty cycle is, of course, per PIN.

vandys, 2017-05-08:
Oh, yes, here's the code I wrote to do a pretty fading light thing with the API.

[colors.txt](https://github.com/micropython/micropython-esp32/files/982048/colors.txt)

dpgeorge, 2017-05-09:
> I'm actually quite surprised by this because I thought I had already implemented it!

We have DAC, but not PWM...

mattytrentini, 2017-05-10:
I see that @vandys has submitted PR #79, just linking here for reference.

dpgeorge, 2017-05-24:
Implemented, see #96

closed 2017-05-24 by dpgeorge