micropython/micropython-esp32 #76: It seems that we can't execute python codes in two different freerots threads!!!

FFtust, 2017-05-05:
The "mp_task" parses and  executes the python file " main.py",it works well.
Now I want to create a new task to execute a python file "factory.py", just like "main.py". my code like this:  

        void mp_main_py_task(void *pvParameter){
             mp_stack_set_top((void*)&pvParameter);
             mp_stack_set_limit(MP_TASK_STACK_SIZE - 512);
             pyexec_file("factory.py");
             vTaskDelete( NULL );  
       }

This new task works well ,but then,when I sent some python code  or just some meaningless code to the cpu by UART, it reboot soon; I found that the configuration"mp_stack_set_top((void*)&pvParameter)"  influenced the python code parsing;
Now my question is:
Using  two thread to parse python codes is allowed?If so ,how can I do it?

dpgeorge, 2017-05-05:
> Using two thread to parse python codes is allowed?

No, it's not allowed the way you have done it because the VM has global state, and 2 (or more tasks) will clash in their use of this state.

> If so ,how can I do it?

Multithreading is supported by the `_thread` module.  If you want to use multithreading at the C level then look at mpthreadport.c and use mp_thread_create to make a new thread.  You'll need to make sure you use the GIL correctly as well.

FFtust, 2017-05-10:
I tried to use the _thread module,and I have saw that my codes have excuted,but later,it rebooted soon;
I configed the MICROPY_PY_THREAD and MICROPY_PY_THREAD_GIL to  1;What did you mean "use the GIL correctly",would I need to do some other configer?
the codes: 

       import _thread
       def thread_entry():
            print("done")
      
      _thread.start_new_thread(thread_entry,(1,))

dpgeorge, 2017-05-10:
> I tried to use the _thread module,and I have saw that my codes have excuted,but later,it rebooted soon;

If you have some Python code that uses the _thread module and has a bug then please file a separate report for that with the code that doesn't work.

> I configed the MICROPY_PY_THREAD and MICROPY_PY_THREAD_GIL to 1

They are already configured to those values, you shouldn't need to change them.

> What did you mean "use the GIL correctly",would I need to do some other configer?

If you want to use threading in C (not  in Python using the _thread module) then you will need to work a bit harder to get things working correctly.  There's nothing extra to configure, but you need to make sure that if you create a new FreeRTOS task that is going to run some Python code then it has acquired the GIL correctly.  The code in py/modthread.c (functions mod_thread_start_new_thread and thread_entry) will help you understand what to do.

FFtust, 2017-05-10:
I have not updated to the newest version,I will try again with the newest files,thank you for your help!

nickzoic, 2017-05-11:
Hey Damien,

    Just out of interest, how practical do you think it'd be to run a
    second Python interpreter on the second CPU, with its own separate
    GIL and the two python interpreters communicating only via message
    passing of one sort or another?
-----Nick

dpgeorge, 2017-05-24:
This is not really a bug report so closing.

closed 2017-05-24 by dpgeorge