micropython/micropython-esp32 #78: Implement machine.UART

oddstr13, 2017-05-07:
`machine.UART` does not seem to be currently implemented, or atleast not exposed Python-side.

Either that, or I missed a setting somewhere / failed the compile.

```python
>>> import machine;dir(machine)
['__name__', 'mem8', 'mem16', 'mem32', 'freq', 'reset', 'idle', 'disable_irq', 'enable_irq', 'time_pulse_us', 'Pin', 'TouchPad', 'ADC', 'DAC', 'I2C', 'SPI']
```

[Build log](http://jenkins.openshell.no/job/Embedded/job/micropython-esp32/7/console).
[Firmware file](http://jenkins.openshell.no/job/Embedded/job/micropython-esp32/7/artifact/artifacts/firmware.bin).

(Please note that this jenkins host is not suited for a large amounts of downloads. If that is desired, it can be arranged.)

dpgeorge, 2017-05-09:
No it's not currently there.  Feel free to have a go at implementing it.

oddstr13, 2017-05-09:
In the past couple of days I've tried implementing it Python side.
https://oddstr13.openshell.no/paste/WRELkHTu/
It's mostly working, except I haven't figured out how to attach UART2 to the IO pads (uart2_enable).

I should probably give implementing it properly a go, but I'm not currently confident in my ability to get the C <-> Py stuff working. On the other hand, I'd get access to the ESP-IDF helper functions then.

dpgeorge, 2017-05-10:
> In the past couple of days I've tried implementing it Python side.

Wow, that looks pretty neat!  You are using the low-level registers to configure it.

> On the other hand, I'd get access to the ESP-IDF helper functions then.

With the esp32 and using ESP IDF it's probably a good idea to go through the IDF helper functions, because there're a lot of details to do with cache, interrupts and mutex's that needs to be considered otherwise the code won't work with the rest of the system; see eg a3fd435d0f0efeec28f003f5d5c002dbee51073b

To get a C implementation of UART you can copy the esp8266/machine_uart.c file as a starting point for the boilerplate code.

nickzoic, 2017-05-10:
(Also, if you get stuck with the wrapper stuff, drop me an email on
nick@zoic.org , I'm happy to help if I can)

oddstr13, 2017-05-11:
There are several issues with my current code, but a basic test works.
```python
import machine
u2=machine.UART(2, baudrate=9600, rx=16, tx=17, timeout=10)
u2.write("Hello, World!\n")
print(u2.readline())
```
Note: `rx` and `tx` need to be set, otherwise the pins won't get connected to the uart.

Should `UART_PARITY_*`, `UART_STOP_BITS_*` and such be exposed under `machine.UART`?

Any comments?

dpgeorge, 2017-05-11:
> Should UART_PARITY_*, UART_STOP_BITS_* and such be exposed under machine.UART?

No, please just use simple integers, following the docs here: http://docs.micropython.org/en/latest/pyboard/library/pyb.UART.html#pyb.UART.init

> Should I make a pull-request, even tho the code isn't up to standard yet?

Yes please.

oddstr13, 2017-05-11:
I might not have been clear enough in my question;
exposed, as in constants - `machine.UART.PARITY_EVEN = 2`, `machine.UART.STOPBITS_1 = 1` and so on.
```python
machine.UART(2, parity=machine.UART.PARITY_NONE)
```

dpgeorge, 2017-05-11:
Setting parity and stop bits should use built-in constants from the user's point of view, eg: `UART(2, parity=None, stop=1)`.  Use "1" for odd parity and "0" for even parity (although behind the scenes you can just check the low bit so that odd numbers are odd parity and even numbers are even parity).

dpgeorge, 2017-05-25:
UART now implemented.

closed 2017-05-25 by dpgeorge

goatchurchprime, 2018-04-18:
This works really well on a TTGO (ESP32) where the UART(2).rx clashes with the OLED screen reset pin.  

But there's no documentation anywhere for setting these rx and tx pins separately that I can find.

steek1337, 2018-04-26:
I try to understand this UART...  I did try use this exempel but nothing happens. Im using  Expansion Board 2,0 / https://docs.pycom.io/chapter/datasheets/downloads/expansion2-pinout.pdf

print('Main')
from machine import UART

uart = UART(1,baudrate= 9600, pins=('P0','P1'))

#P0 = tx
#P1 = rx

print('Hello World')
#uart.write('Hello')

#uart.read(5)

print(uart.any())
print(uart.write('123'))
print(uart.write(b'abcd'))

print(uart.any())

robert-hh, 2018-04-26:
You are using the pins of UART0 assigned to UART1. However, UART0 is used for the REPL prompt. The default pins for UART1 are P3 and P4.
Besides that, questions about the Pycom port and Pycom devices are better asked at the Pycom forum:
https://forum.pycon.io

MrSurly, 2018-04-26:
@steek1337 

Firstly, you're using Pycom Micropython, which is a derivative of this (mainline) Micropython.

Assuming you're looking to see data on a serial port on your computer, using the expansion board:

* Ensure the expansion board jumpers for serial are installed
* The normal pinout for a WiPy board is that P0 = RX, and P1 = TX; your example is the opposite, and likely the reason it's not working.  The "P0 FTDI_TX" and "P1 FTDI_RX" on the expansion board pinout are correct -- but from the point of view on the computer, hence the "FTDI," which is referring to the USB->Serial chip.

arvabhatrajesh, 2019-10-22:
is it multi threading possible in esp32 on micropython ide ...
is it possible my uart multithreading nit working can anyone suggest me for perfect
***MULTITHREADING in uart and led***

import _thread as thread
import time
from machine import Pin
import machine
import esp32
import micropython
from machine import UART

led25 = Pin(25, Pin.OUT)
led26 = Pin(26, Pin.OUT)

uart=machine.UART(1, baudrate=115200, rx=16, tx=17,timeout=50000)
uart.write("Hello,Hello,Hello!!!!!!\n")

def loop_led25(e):
  while True:
    led25.value(not led25.value())
    time.sleep(e)

def loop_led26(e):
  while True:
    led26.value(not led26.value())
    time.sleep(e)
  
def loop_uart(e):
  while True:
    print(uart.readline())
    time.sleep(e)
    
thread.start_new_thread(loop_led25,(.5,))
thread.start_new_thread(loop_led26,(.5,))
thread.start_new_thread(loop_uart,(2,))

code is dumped to my esp32 but multithreading is not done

dpgeorge, 2019-10-22:
@arvabhatrajesh for questions please ask them at https://forum.micropython.org