micropython/micropython-esp32 #82: SHA hardware support

vandys, 2017-05-10:
Is there any reason not to use the HW SHA support (and possibly AES)?  They are in the SDK
and should get the job done with smaller code size (and no doubt they run faster).

dpgeorge, 2017-05-10:
The only reason they aren't used at the moment is because no one has yet got round to enabling them and testing them.  So feel free to go for that!

vandys, 2017-05-11:
Ok, I've got the code running.  Still needs to be tested, but I'm seeing about a 40% speed
improvement with the ESP32's SHA support.  Is that worth pursuing in your mind?

nickzoic, 2017-05-11:
Hell yeah :-)  Thanks for looking into this!

-----N
--
Nick Moore <nick@zoic.org> 0409 656 267

vandys, 2017-05-11:
Ok, I'll dig up some regression testing cases, probably turn back on SHA1 for testing just in case
somebody builds that mode.  You should see a pull request in the next week or so, thanks!

vandys, 2017-05-11:
Code for review: https://github.com/micropython/micropython-esp32/pull/84

vandys, 2017-05-11:
Oh yes, code lineage.  This pull request is just a copy of the extmod/uhashlib.c with calls to the
software SHA replaced with calls to the mbedtls ones (which use the CPU's SHA support).
That plus a few minor changes to storage allocation.

dpgeorge, 2017-05-24:
Merged, see #95

closed 2017-05-24 by dpgeorge

saramonteiro, 2022-02-03:
I know this issue is quite old. But let me ask:
Does hashlib rely on the top of HW support or is it based on a software implementação?
Do we have HW SHA 512 support for ESP32 in upython?

vandys, 2022-02-03:
It uses HW support.  The chip does SHA512, but I'm pretty sure I didn't support that.  Have a look at moduhashlib for the esp32 port to see if somebody added it after my time.

saramonteiro, 2022-02-03:
> It uses HW support. The chip does SHA512, but I'm pretty sure I didn't support that. Have a look at moduhashlib for the esp32 port to see if somebody added it after my time.

Thanks for the answer. Where can I take a look on that?

vandys, 2022-02-03:
Check out a Micropython source tree from github, then look at extmod/moduhashlib.c

Which I just did, and it has SHA1 and SHA256, but not SHA512.  So I guess you can request that as an enhancement, which would entail (1) convincing the devs it's a fit for Micropython, and then adding a SW version plus HW hooks to chips like the ESP32 which have hardware assist.