micropython/micropython-esp32 #87: Setting the time on an esp32

rem101, 2017-05-13:
I've worked with the ESP8266, and there I would set the time on bootup via ntp.  I was able to call the RTC class from machine and also, use utime.localtime to help calculate the delta.  Then I would call RTC().datetime to set the time accordingly.  In the ESP32 port, there is neither an RTC class or a localtime function from utime, I'd be glad to look at adding it, but I wanted to make sure I'm not duplicating the same functionality elsewhere.  Is someone already working on something similar?

dpgeorge, 2017-05-14:
> I wanted to make sure I'm not duplicating the same functionality elsewhere. Is someone already working on something similar?

Not to my knowledge, so feel free to go for it.

mcauser, 2017-05-16:
Some work on machine.RTC here: https://github.com/micropython/micropython-esp32/pull/85

rem101, 2017-05-17:
Thanks mcauser!  I'll just add the localtime method to utime to complete setting the time.

nickzoic, 2017-05-17:
Cool, I should add a call to `machine.RTC().init()` to mpy-utils as well ...

MrSurly, 2017-05-17:
Actually, I did implement this in https://github.com/micropython/micropython-esp32/pull/85

Though, I need to update it to make a bit more sense.

vandys, 2017-05-22:
Has anybody wired in the SNTP code form the SDK?  Seems like it'd be a nice fit.

rem101, 2017-06-11:
Thanks @MrSurly, that's what I needed to set the time from ntp.

closed 2017-06-11 by rem101