micropython/micropython-esp32 #88: Catching up with SDK?

vandys, 2017-05-14:
Micropython doesn't build with the latest SDK bits; what's your philosophy for moving the "working"
version of the SDK forward?

dpgeorge, 2017-05-14:
> what's your philosophy for moving the "working" version of the SDK forward?

Apart from any regressions that might be introduced in the upstream SDK there's no reason not to update continuously.  Of course, someone needs to have time to do it :)  When doing it the main thing to check is for changes to the `components/esp32/ld/esp32.common.ld` file (eg via `git log components/esp32/ld/esp32.common.ld`) which should be mirrored to the local `esp32.custom_common.ld`

nickzoic, 2017-05-15:
In terms of philosophies, so far it's been "whenever we get the chance"!
Progress in the ESP-IDF has been quite rapid. PRs to catch up to the
latest IDF are welcomed ... mostly these are pretty trivial changes to
headers and so on.
I think now we've hit IDF v2.0 we should try to minimize the chaos by
sticking to IDF releases for the main 'esp32' branch ... obviously if
we're developing something which needs new IDF features, that
development branch will need the newer IDF but we can then wait for the
feature to hit an IDF release before merging the MicroPython branch back
into 'esp32'.  That way breaking API changes become much rarer.
This may of course prove to be *too* cautious, but I think it'd help to
come up with a strategy and see how it turns out.
-----N

MrSurly, 2017-05-15:
Coincidentally, I was doing this today: https://github.com/micropython/micropython-esp32/pull/89

I needed to get around this already-fixed issue in the IDF: https://github.com/espressif/esp-idf/issues/520

MrSurly, 2017-05-15:
Which also leads to the question: Can the tests be made to run on the ESP32?  Or does it only work on the pyb?

dpgeorge, 2017-05-16:
> Can the tests be made to run on the ESP32?

The test suite runs on the esp32.  You can use esp8266 target and ignore ones that seem to be esp8266 specific.  There are no network tests, that's a TODO.

MrSurly, 2017-05-17:
@vandys Should be (relatively) up-to-date, now

vandys, 2017-05-18:
Yes, I've updated SDK and tree, built and run.  Also did some networking, and everything looks fine.  Many thanks (I was planning on doing it myself :->)!

closed 2017-05-18 by vandys