micropython/micropython-esp32 #90: OSError: [Errno 2] ENOENT

mcauser, 2017-05-16:
Installed the latest micropython-esp32 on a new [Wemos Lolin32](https://www.aliexpress.com/store/product/WEMOS-LOLIN32-V1-0-0-wifi-bluetooth-board-based-ESP-32-4MB-FLASH/1331105_32808551116.html) board.

On first boot, I noticed there was an `ENOENT` error:

```
>>> 
PYB: soft reboot
OSError: [Errno 2] ENOENT
MicroPython v1.8.6-1089-g1365f7f-dirty on 2017-05-16; ESP32 module with ESP32
Type "help()" for more information.
```

I thought the issue may have been baud related, so I tried 115200, 230400, 460800 and 921600.
Tried importing _boot to re-setup the /flash, but kept getting "FAT filesystem appears to be corrupted."
Tried using a different machine and a few different usb cables too.
Then I tried manually creating the main.py file, and the problem went away.

After `make erase` and `make deploy`, there appears to be no main.py file in /flash.
Should there be an empty / placeholder file, like with boot.py?

```
>>> import os
>>> os.listdir()
['boot.py']
```

So, I created a main.py:

```
>>> f = open("main.py", "w")
>>> f.write("print(\"main.py: Hello\")\n")
24
>>> f.close()
>>> os.listdir()
['boot.py', 'main.py']
```

And on soft and/or hard reboot, no more `ENOENT` error!

```
>>> 
PYB: soft reboot
main.py: Hello
MicroPython v1.8.6-1089-g1365f7f-dirty on 2017-05-16; ESP32 module with ESP32
Type "help()" for more information.
>>>
```

```
>>> ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3132
ho 0 tail 12 room 4
load:0x40078000,len:7788
load:0x40080000,len:252
entry 0x40080034
I (1321) cpu_start: Pro cpu up.
I (1321) cpu_start: Single core mode
I (1322) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1335) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1355) heap_alloc_caps: At 3FFD48C8 len 0000B738 (45 KiB): DRAM
I (1376) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1398) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1419) heap_alloc_caps: At 400998C8 len 00006738 (25 KiB): IRAM
I (1440) cpu_start: Pro cpu start user code
I (1501) cpu_start: Starting scheduler on PRO CPU.
main.py: Hello
MicroPython v1.8.6-1089-g1365f7f-dirty on 2017-05-16; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```

FYI, version is showing dirty as I modified the Makefile with:

```
ESPIDF = /Users/mike/esp/esp-idf
PORT = /dev/tty.SLAB_USBtoUART
BAUD = 230400
```

mcauser, 2017-05-17:
On the ESP8266, the error is a little more user friendly:
```
PYB: soft reboot
#5 ets_task(40100164, 3, 3fff8388, 4)
could not open file 'main.py' for reading
MicroPython v1.8.7-75-gd2b2a44 on 2017-02-02; ESP module with ESP8266
Type "help()" for more information.
```

pfalcon, 2017-05-17:
There's the same ENOENT error with current esp8266 builds.

dpgeorge, 2017-05-17:
The old "could not open file ..." was replaced with the (proper) OSError(ENOENT) message.  It's not harmful, but it is perhaps confusing.

closed 2017-05-18 by mcauser

mcauser, 2017-05-18:
Confirmed, same `OSError: [Errno 2] ENOENT` message now appears on upgraded ESP8266.

dss91, 2018-06-29:
i still face the same error on my node mcu board and further i am not able to type in REPL prompt, is the issue still there ? i am not sure what is wrong

dss91, 2018-06-29:
i tried to read debug messages on startup at high baud rate , which showed up
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31108, room 16
tail 4
chksum 0x28
load 0x3ffe8000, len 1100, room 4
tail 8
chksum 0x4e
load 0x3ffe8450, len 3268, room 0
tail 4
chksum 0x09
csum 0x09
#0 ets_task(4020a9c0, 31, 3ffe9170, 4)
#1 ets_task(4021e890, 32, 3ffeabe0, 34)
#2 ets_task(4021d8c4, 20, 3ffea940, 3)
rf cal sector: 1019
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.2.0-dev(9422289) compiled @ Nov  3 2017 19:40:08
phy ver: 1136_0, pp ver: 10.2

mode : softAP(6a:c6:3a:ab:56:04)
add if1
#3 et)▒*▒▒+Z▒▒▒Q▒▒▒▒▒UU▒U)B▒*ŢT▒▒▒▒ZuV▒j▒R*U▒ZT▒TT▒▒▒Q▒UTU▒▒▒T▒▒uT▒▒m*▒PT▒▒UUE▒X
                                                                               ▒                                                                                        UT▒U▒uZU▒▒U▒▒▒▒▒▒▒▒tס

robert-hh, 2018-06-29:
This repository is closed and the development for ESP32 is continued at the main mp repository https://github.com/micropython/micropython with pre-built images at http://micropython.org/download#esp32. Mayby it's better to try that version.

bhcuong2008, 2021-06-10:
I have faced this issue. I test it from RAM disk. I have asked in the forum. Hope that it could be solved here.

https://forum.micropython.org/viewtopic.php?f=2&t=10655

Thank you.

bhcuong2008, 2021-06-11:
After traceback, I found the problem. It needs to enable MICROPY_READER_VFS. It solved now.

DamirKh, 2021-09-13:
> After traceback, I found the problem. It needs to enable MICROPY_READER_VFS. It solved now.

Did You mean micropython/ports/esp32/mpconfigport.h:36 ?

bhcuong2008, 2021-09-14:
@DamirKh , yes. 
It's `#define MICROPY_READER_VFS                  (1)`