micropython/micropython-esp32 #91: Pre-built binaries

anteph, 2017-05-17:
Hi,

Any chance that pre-built binaries are made available? Something similar to the ESP8266 micropython:
http://micropython.org/download#esp8266, where we just need to upload the binary with the esptool.

So people with less experience don't need to set the whole toolchain in order to start playing with micropython. 

Best regards,
Nuno Santos

mcauser, 2017-05-18:
Here's some simplified instructions for building and deploying on OSX.
https://forum.wemos.cc/topic/380/micropython-lolin32
Took me more time to write instruction than it did to actually run.

It's as simple as, download a usb driver, clone 3 repos, make a few text changes, run a few makes.

But yes, precompiled public binaries would be nice.

mattytrentini, 2017-05-18:
I've been toying with the idea of putting in place a Continuous Integration system that could do this; maybe it would be worthwhile?

dpgeorge, 2017-05-19:
Firmware now available here: https://micropython.org/download/#esp32

closed 2017-05-19 by dpgeorge

anteph, 2017-05-20:
@dpgeorge thank you very much for making the binaries available :). 

I've just tested uploading them with the esptool and it is working fine. I've written a small post about the procedure needed to flash it:
https://techtutorialsx.com/2017/05/20/esp32-micropython-support/

@mattytrentini sounds interesting, how would it work?

anteph, 2017-06-19:
Hi! @dpgeorge , the firmware is no longer available at the https://micropython.org/download/#esp32 link. Was it moved to other location or just temporarily missing?

Thanks,
Nuno Santos

dpgeorge, 2017-06-20:
@anteph thanks for the heads-up, should now be fixed (but may take a day to update itself properly).

nickzoic, 2017-06-20:
Hi Damien, did we conclude in #109 that we were going to need different
binary builds for different memory sizes due to the changes in the
bootloader?   --Nick

dpgeorge, 2017-06-20:
> did we conclude in #109 that we were going to need different binary builds for different memory sizes due to the changes in the bootloader

There was no concrete conclusion... probably this point needs clarifying from upstream (ie Espressif).  It may be that we can get away with providing firmware for 4mb modules, and that will work well enough for other modules.  We anyway require at least 4mb (since the firmware is >1mb, the filesystem starts at 2mb).  And probably firmware configured for 4mb will work on hardware with 8mb or more flash (but you just won't be able to use the space beyond 4mb for storage).

rdagger, 2017-07-03:
@dpgeorge looks like the [firmware](http://micropython.org/download#esp32) isn't available again.

dpgeorge, 2017-07-04:
@rdagger thanks for the heads up.  I've improved the auto-build scripts so they automatically update the ESP IDF and it should now be fixed.

jpvic, 2017-09-06:
@dpgeorge ,

sorry to say but there is no [firmware](https://micropython.org/download/#esp32) available again .

And I can't find the esp32 diretory  since the dir layout is applied : https://github.com/micropython/micropython/tree/master/ports

If I can recall correctly , the manual for building the esp32 firmware is disappeared .

robert-hh, 2017-09-07:
The esp32 directory is at https://github.com/micropython/micropython-esp32/tree/esp32/ports below https://github.com/micropython/micropython-esp32. In the ports/esp32 subdirectory you'll find also the README.md with building instructions.

dpgeorge, 2017-09-07:
Thanks @jpvic for the report.  The issue was that the auto-build scripts were not updated to reflect the move of the ports to the "ports/" subdirectory.  It's now fixed.

jpvic, 2017-09-07:
@robert-hh  ,
Thank you . I will try out the instructions. But the directory ESP32 wasn't there at the moment of posting my message.

@dpgeorge ,
I see that the version 1.9.2. is also arrived. Temporaly, 1.9.1 was visible. 

Issue is solved.

Thank you both!

chrisb1010, 2017-10-26:
@dpgeorge, 

To plagiarize a previous comment, sorry to say but there is no firmware available again.

Any chance to have it available again?

dpgeorge, 2017-10-27:
@chrisb1010 I don't see any problems, the esp32 firmware is there, right now it's called: esp32-20171027-v1.9.2-283-gdba3796e.bin

DavidAntliff, 2017-12-21:
Is this the "no firmware available again" symptom? Is it meant to show some files for the ESP32?

[image]

dpgeorge, 2017-12-22:
It's there again!