micropython/micropython-esp32 #94: Implement MCPWM module support

MrSurly, 2017-05-19:
I'm going to begin [development](https://github.com/MrSurly/micropython-esp32/tree/dev-mcpwm) on a thin wrapper around [MCPWM](http://esp-idf.readthedocs.io/en/latest/api/peripherals/mcpwm.html?highlight=mcpwm), starting with implementation of capture.

eugenio, 2017-06-07:
The MCPWM link is dead... new link is [MCPWM](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/mcpwm.html)

MrSurly, 2017-06-07:
Updated URL.

nickzoic, 2017-08-03:
Sounds like a great idea.  Someone earlier (LinuxConfAU, I think) brought up the idea of using the MCPWM "RMT" function to clock bits out to NeoPixel type LEDs, which seems like it might work too.

bskp, 2019-09-24:
Since I needed 6 PWM channels with independent frequencies, I extended the **esp32-module** with **MCPWM** support: https://github.com/bskp/micropython_esp32_mcpwm.
Currently, only basic functionality is covered and I do not plan to dive much deeper. However, I'm happy if it helps someone else!

bskp, 2019-09-24:
@dpgeorge thank you for the notice.
Is there a chance for this feature being merged upstream, even though the API deviates from the one of machine.PWM? If so, I'll open a pull request.

nickzoic, 2019-09-24:
G'day @bksp for what it's worth I *still* think this is a great hardware feature to have exposed in MicroPython :-)

The work in this repo has now been merged back into the main micropython/micropython repo, and it'd be well worth opening this issue over there and working towards an agreement on an API change to allow support for this.  I'm happy to help if I can.

@damien did we work out how to lock this repo to prevent confusion?

dpgeorge, 2019-09-25:
> did we work out how to lock this repo to prevent confusion?

It's possible to archive a repository to make it read-only, but I think it's good to keep this un-archived so we can close/update issues/PRs here as they are resolved upstream.  (And we rarely get comments on this repo anymore anyway.)