micropython/micropython-esp32 #102: WiPy 2.0 Wifi not working

garaged, 2017-05-27:
I updated a WiPy 2.0 with esp32 firmware a couple of times now (just updated) and when I try to connect wifi I get this endless loop of error messages:

I (679386) wifi: STA_DISCONNECTED, reason:2
I (681016) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (681016) wifi: state: init -> auth (b0)
I (682016) wifi: state: auth -> init (2)
I (682016) wifi: STA_DISCONNECTED, reason:2
I (682886) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (682896) wifi: state: init -> auth (b0)
I (683896) wifi: state: auth -> init (2)
I (683896) wifi: STA_DISCONNECTED, reason:2
I (684316) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (684316) wifi: state: init -> auth (b0)
I (685316) wifi: state: auth -> init (2)
I (685316) wifi: STA_DISCONNECTED, reason:2
I (687096) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (687096) wifi: state: init -> auth (b0)
I (688096) wifi: state: auth -> init (2)

Wifi worked for a couple of minutes yesterday, it disconnected and after that I haven't been able to connect again, even using wlan.scan().

Actually, scan() sometimes fails with a weird message:

>>> wlan.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'wlan' is not defined

And other times it works as expected.

It may be related, but I can't find a way to debug this problem, I cant find what "reason:2" means in the connection attempt.

my current build is:
MicroPython v1.8.7-962-g6693f91f-dirty on 2017-05-27; ESP32 module with ESP32

dpgeorge, 2017-05-28:
WiPy and LoPy boards need to have the antenna selected via a GPIO pin before the wifi works.  Please see the end of the esp32/README.md file for details.

garaged, 2017-05-28:
Confirmed, it worked after I did that, I remember reading that before, but never came back to it because WIFI worked for a few minutes without selecting antena, it stoped working after some 3-5 minutes (I know because I was registering a DHT22 module in my thingsboard).

Thanks a lot George ! sorry for my mistake in this.

closed 2017-05-28 by garaged