micropython/micropython-esp32 #106: Hall sensor don't  work!

FFtust, 2017-05-31:
I got the value of hall sensor like this:
         
        adc1_config_width(ADC_WIDTH_12Bit);
        value=hall_sensor_read();

but  the return value don't change much when I take a  magnet close to it .
I viewed the document about esp32, it shows that ADC_CHANNEL_0 &  ADC_CHANNEL_3 are combined to  hall sensor, for hardware, are GPIO36 and GPIO39. Is it true that I can't use these two pins if I want to use hall semsor?  In another words, need I make these two pins not connect to anything?

MrSurly, 2017-05-31:
Could you post a complete code snippet that can be run?  That would be helpful.

dpgeorge, 2017-06-01:
Are you using C or (Micro)Python to write your code?  Please note that this is the repository for MicroPython running on the ESP32, not general support for using the ESP32.

dpgeorge, 2017-07-26:
Closing due to inactivity.

closed 2017-07-26 by dpgeorge