micropython/micropython-esp32 #107: Question: what's necessary to make webrepl_setup work on esp32 port?

eugenio, 2017-06-02:
As per question what are the necessary requirements and effort necessary?
Is there an expected date for it?
Can I help somehow? I do know how to code somehow.

dpgeorge, 2017-06-03:
The way webrepl is implemented on esp8266 prevents it from being easily ported to esp32.  In particular the esp8266 implementation relies on custom behaviour of lwIP whereby one can register a callback for events like accept and read.

To make webrepl run as a daemon (in the background) on esp32 would require either creating a special task/thread for it to run in (and then implementing webrepl all in C), or adding a dedicated hook to run periodically in the main uPy task in order to check for incoming connections.

eugenio, 2017-06-08:
I'm sorry but this is above my knowledge... is there any other way i can contribute?

nickzoic, 2017-06-17:
(I'm having a go at this, by the way ...)

mitchins, 2017-09-05:
Is this specifically a feature lacking on the ESP32, or do we simply not have an API for it?

dpgeorge, 2017-09-06:
> Is this specifically a feature lacking on the ESP32

WebREPL is a feature that is lacking, and requires a bit of effort to implement.

mitchins, 2017-09-06:
Thanks, but I was actually curious if it’s the WS send/receive hook methods that were missing or undiscovered as that’s what seems to be the key to 8266 implementation that’s missing.

Seems like otherwise, a thread or routine to service Telnet might be the alternative.

dpgeorge, 2017-09-06:
> if it’s the WS send/receive hook methods that were missing or undiscovered

They don't exist on the esp32 because this port uses the high-level lwIP interface (like socket_open) whereas esp8266 uses the low-level lwIP interface and therefore can implement callbacks for connections and received data.

nickzoic, 2017-09-06:
> Seems like otherwise, a thread or routine to service Telnet might be
> the alternative.
Yeah, I implemented this at
https://github.com/nickzoic/micropython-esp32/tree/esp32-rrepland it works but I was considering getting tinyssh library going
instead.

mitchins, 2017-09-06:
Interesting thanks guys.

I’m happy to spend some time contributing...

From what I see, there’s one question:
Is it possible to get the low level call access like ESP8266 with extra work and focus the efforts on that? - or should the effort be focused on developing alternatives to Webrepl.

@nick, telnet might be simpler and have less requirements?

dpgeorge, 2017-09-06:
> Is it possible to get the low level call access like ESP8266 with extra work and focus the efforts on that?

It's difficult to do that because of FreeRTOS and threads and reentrancy of lwIP.  It would be better to find an alternative way to support WebREPL.

cefn, 2018-03-21:
Has there been any progress on this? What can be usefully attempted to get closer to working Webrepl on ESP32?

Lab-8916100448256, 2018-04-18:
Hi,

@cefn,  I'm also interested in this feature and here is what I found so far :

There is a typo in the URL that @nickzoic gave for his telnet REPL code. You need to remove "and" at the end : https://github.com/nickzoic/micropython-esp32/tree/esp32-rrepl

I found another implementation of remote REPL though Telnet for the esp32 in this repository : https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo

dpgeorge, 2018-04-27:
WebREPL is now supported on the ESP32!  It works pretty much exactly the same as the ESP8266.  See commit in the main repo: https://github.com/micropython/micropython/commit/23e9c3bca7eb40566ffd4b6cd57b9b4e1f19ad30

> To make webrepl run as a daemon (in the background) on esp32 would require either creating a special task/thread for it to run in (and then implementing webrepl all in C), or adding a dedicated hook to run periodically in the main uPy task in order to check for incoming connections.

The latter approach was taken.

closed 2018-04-27 by dpgeorge