micropython/micropython-esp32 #110: machine.reset() results in WDT reset

MrSurly, 2017-06-13:
Issuing a `machine.reset()` results in a slew of `IllegalInstruction` errors followed by a WDT timeout.

I'm trying to implement `machine.reset_cause()`, but this problem means that whenever you use `machine.reset()`, `machine.reset_cause()` will return `machine.WDT_RESET`

Code is here: https://github.com/MrSurly/micropython-esp32/tree/dev-reset-cause, branched from mainline ESP32.  Note that `machine.reset()` has _not_ been  modified.

I guess my question is: Should this be an issue for the IDF?  Or is it a problem in µPy?

```
MicroPython v1.8.7-963-ge68bf91-dirty on 2017-06-13; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> machine.reset()
E (91004) wifi: esp_wifi_stop 802 wifi is not init
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
-----8<----------- < SNIP, many repeats removed > ------------------
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Guru Meditation Error of type I�ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3416
load:0x40078000,len:9492
load:0x40080000,len:252
entry 0x40080034
I (1275) cpu_start: Pro cpu up.
I (1275) cpu_start: Single core mode
I (1277) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1290) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1310) heap_alloc_caps: At 3FFD5290 len 0000AD70 (43 KiB): DRAM
I (1331) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1352) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1374) heap_alloc_caps: At 400929E8 len 0000D618 (53 KiB): IRAM
I (1395) cpu_start: Pro cpu start user code
I (1454) cpu_start: Starting scheduler on PRO CPU.
OSError: [Errno 2] ENOENT
MicroPython v1.8.7-963-ge68bf91-dirty on 2017-06-13; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```

MrSurly, 2017-06-13:
One more note -- made a quick test app using just the IDF -- it resets normally.

dpgeorge, 2017-06-14:
> One more note -- made a quick test app using just the IDF -- it resets normally.

Did the test app enable the CONFIG_WIFI_ENABLED option?  That is enabled by uPy and might be the cause of the issue.

MrSurly, 2017-06-14:
Yes.  It's the standard template:
https://github.com/espressif/esp-idf-template.git

I only modified it to add my own WiFi credentials, and made it call
`esp_restart()` after 10 seconds.  It works cleanly with both current IDF
master, and the ESP32 IDF detached head.

MrSurly, 2017-06-14:
Yes.  It's the standard template: https://github.com/espressif/esp-idf-template.git, which is a WiFi demo.

I only modified it to add my own WiFi credentials, and made it call `esp_restart()` after 10 seconds.  It works cleanly with both current IDF master, and the ESP32 IDF detached head.

dpgeorge, 2017-07-31:
Revisiting this issue after many IDF updates, reset() seems to work better, but still not correctly: it doesn't have any IllegalInstruction's, instead it just hangs for a bit then there's a WDT reset.

dpgeorge, 2019-01-16:
machine.reset() works with latest upstream code.

closed 2019-01-16 by dpgeorge