micropython/micropython-esp32 #111: nghttp2_*: No such file or directory

pacmac, 2017-06-13:
Just did a clean clone of the current version and the make results in:

	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_http.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_version.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_mem.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_hd_huffman.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_rcbuf.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_callbacks.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_session.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_stream.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_hd.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_priority_spec.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_buf.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_option.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_npn.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_helper.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_frame.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_outbound_item.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_hd_huffman_data.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_pq.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_queue.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_submit.o: No such file or directory
	xtensa-esp32-elf-ld: cannot find build/esp-idf/components/nghttp/library/nghttp2_map.o: No such file or directory

dpgeorge, 2017-06-14:
Did you do "git submodule update --init" in the IDF directory?

dpgeorge, 2017-06-28:
Assuming fixed.

closed 2017-06-28 by dpgeorge