micropython/micropython-esp32 #114: UART.write loses 2 bytes

rdagger, 2017-06-22:
After I initialize UART2, the first 2 bytes written disappear. After the 2 lost bytes, it works as expected.  For example with the following code only the last 2 bytes 0x03 and 0x04 are transmitted:

```
from machine import UART
uart = UART(2, 9600)
uart.write(bytes([0x01, 0x02, 0x03, 0x04]))
```
Issuing the write statement a 2nd time transmits all 4 bytes.

dpgeorge, 2017-06-22:
Yes, I saw this bug as well.  I think it's a bug in the Espressif IDF and I posted an issue about it: https://github.com/espressif/esp-idf/issues/710

In the meantime, if you really need to have it fixed, you can disable the TX buffer in the UART by changing the uart_driver_install line to (tho 32 becomes a 0):
```
uart_driver_install(uart_num, 256, 0, 10, &UART_QUEUE[self->uart_num], 0);
```

rdagger, 2017-06-22:
Thanks, only the first write statement is affected.  My work-around is to call
`uart.write(bytes([0x00]))`
after the init statement.  Subsequent write statements work correctly.

dpgeorge, 2017-06-28:
This is fixed in 965027d7c9595cdc265f3b5110e78b2a50312178, where the TX buffer is increased to 64 bytes.

closed 2017-06-28 by dpgeorge