micropython/micropython-esp32 #116: Can not build

DaneGrinder, 2017-06-25:
Hi

I can't build this anymore. Getting:

machine_dac.c: In function 'mdac_make_new':
machine_dac.c:61:21: error: implicit declaration of function 'dac_output_enable' [-Werror=implicit-function-declaration]
     esp_err_t err = dac_output_enable(self->dac_id);
                     ^
machine_dac.c:63:15: error: implicit declaration of function 'dac_output_voltage' [-Werror=implicit-function-declaration]
         err = dac_output_voltage(self->dac_id, 0);
               ^
cc1: all warnings being treated as errors
make: *** [build/machine_dac.o] Error 1

Worked when I build it a few weeks ago, but now that I updated the repo, and try to build again, this error comes up. Tried deleting the whole build-chain and start over, but no luck.

Hope someone can help me

robert-hh, 2017-06-25:
Did you upgrade a) esp-idf too (better delete it & re-clone it, and b) issued the command "git submodule updata --init" in your esp directory?

DaneGrinder, 2017-06-25:
Believe I did, but doing it again now. Will get back to you in 2 min

DaneGrinder, 2017-06-25:
It worked :-)

I was like 99% sure I already tried this twice, but apparently not

Thanks a lot robert

closed 2017-06-25 by DaneGrinder