micropython/micropython-esp32 #117: uasyncio missing

MissingNoIOI, 2017-06-25:
I've tried downloading the latest build and building it from source, but none contain uasyncio.
```
>>> import uasyncio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'uasyncio'
```
Is there a special build flag or isn't it supported on this board?

dpgeorge, 2017-06-26:
uasyncio is not included by default in the build.  You can install it in 3 ways:
- run `import upip` then `upip.install('micropython-uasyncio')` at the REPL on the ESP32 (requires an internet connection on the board)
- copy the library from your PC (from micropython-lib repo) to the local filesystem on the ESP32
- symlink the library to the esp32/modules/ directory and rebuild the firmware

The easiest thing to do is use upip.

(Note that you'll need the latest build as of e13e9f557429f645dae47392d0c3778ed3f9046b because I enabled the utimeq module which is required for uasyncio to work.)

MissingNoIOI, 2017-06-27:
Ah, thanks, it was my misunderstanding then

closed 2017-06-27 by MissingNoIOI

Sunrise17, 2018-02-11:
Dear @dpgeorge , i have successfully installed 'micropython-uasyncio' lib to my ESP32 board with pip method but now i am receiving following error below; Why can it caused by?

`Traceback (most recent call last):
  File "main.py", line 151, in <module>
  File "/lib/uasyncio/core.py", line 224, in get_event_loop
  File "/lib/uasyncio/__init__.py", line 21, in __init__
  File "/lib/uasyncio/core.py", line 30, in __init__
AttributeError: 'module' object has no attribute 'deque'
MicroPython v1.9.3-286-gbbb08431 on 2018-02-09; ESP32 module with ESP32`