micropython/micropython-esp32 #121: APA102 Driver support

Hokyjack, 2017-07-04:
Hello, I have MicroPython v1.9.1-219-g3580284e installed at my ESP32 NodeMCU development board, but unfortunately I can't find APA102 library included altought it should be in ESP8266 1.9.1 microPython.
```
 from apa102 import APA102
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'apa102'
```
how should I add it so I can use it with ESP32?
and one more question - is this implementation of apa102 using real HSPI (hardware SPI) or it just bit-bangs the bits?

dpgeorge, 2017-07-05:
The APA102 driver for esp8266 is very specific to that chip (it uses inline assembler and has specific timing).  So it's not straightforward to port it to the esp32.  It's anyway just a bit-banging implementation.

It would be easier to write a new driver from scratch for the esp32, in Python, using the SPI peripheral (SCK and MOSI lines).

Hokyjack, 2017-07-05:
Oh, thanks for the answer, I'll try to write my own driver then

mattytrentini, 2017-08-12:
Also see #159

mattytrentini, 2017-09-03:
I'm just wondering why the APA102 implementation was implemented with bit-banging on the ESP8266? I've read the conversation on [PR1941](https://github.com/micropython/micropython/pull/1941) but, while there was some solid, animated discussion and excellent collaboration, I'm not sure why an SPI implementation was chosen? 

Was it perhaps that hardware SPI wasn't available at the time? Or that bit-banging produced faster updates that the 80MHz SPI limit? I'm sure I'm missing something obvious!

My understanding was that it should be possible to write a pure Python APA102 driver that just takes in a `machine.SPI` object. This will work across all ports and be optimised where hardware SPI is available...

mattytrentini, 2017-09-03:
Oh, and @Hokyjack, are you still looking at creating an APA102 driver? If so, you should also be aware of the end-frame issue in the ESP8266 implementation, documented in [PR3037: APA102 LED "n+1" is lighted on write()](https://github.com/micropython/micropython/issues/3037).

mcauser, 2017-09-05:
Mention me (@mcauser) if you want me to test any code on my ESP32 (lolin32) and APA102 strip.

dpgeorge, 2017-09-05:
> My understanding was that it should be possible to write a pure Python APA102 driver that just takes in a machine.SPI object.

Yes it should work, even at low SCK frequency.  But note that the SPI will use up 3 pins on the MCU whereas the APA102 only needs 2.

mattytrentini, 2019-01-13:
With the [micropython-dotstar library](https://github.com/mattytrentini/micropython-dotstar) we can probably close this ticket? @mcauser if you could test that library I'd really appreciate it!

Also [posted on the forum](https://forum.micropython.org/viewtopic.php?f=15&t=5810), for reference.

RobertJBabb, 2019-01-29:
This is probably only of historical interest now, but I developed a MicroPython APA102/DotStar support library as well over the past few weeks: https://github.com/RobertJBabb/micropython_generic_apa102  
The library supports the Adafruit DotStar API, the MicroPython ESP8266 APA102 API, and I added support for Matt's MicroPython SPI DotStar API after I saw his post. I focused on backward compatibilty instead of performance, so I expect that my class will be slower and consume more memory than Matt's. The repository also includes two test files that contain more compact examples of common DotStar demo code than I have seen elsewhere, perhaps that may be of interest to some.
I like Matt's implementation so I will probably use his library going forward. 
As a newbie, what is the best way to coordinate development work with other MicroPython contributors? 
I enjoyed writing this library but I would like to avoid duplicating effort in the future.

dpgeorge, 2019-01-31:
> As a newbie, what is the best way to coordinate development work with other MicroPython contributors?

It's best to start by discussing on the https://forum.micropython.org

mattytrentini, 2019-02-04:
@RobertJBabb As Damien mentioned, raising a topic in the forum is the best bet. Sorry I hadn't stumbled on your effort but I'll be sure to take a look at your implementation!

@BEMNETADEFIS To allow the APA102 driver to work on Linux you just need to come up with a Python class that can satisfy the SPI requirements of the APA102 driver. As mentioned, that means you just need a `write` and a `deinit` (and even the latter is not strictly necessary if you don't ever return the resources). Let me know if you need more assistance.

BEMNETADEFIS, 2019-02-08:
> do i need to develop library
> I do have an experiance of developing python libraries before
> how can i create this
> Recommed me for bigger challange

the more micropython is avaliable for liunx and hacker zones it become more
innovative that why i rised such questions for me linux have python plagin
including
Apt-get update python.org python 5 #
/4/3 that can help for more progressive progrsmming that why
i know this question is certainly difficult but does it fits for the
exsiting inapplicability of micropython in linux
Single board computer for linux
linux devices and much related techologies become an accessible
So it do huge challange and have numerious opportunities
Regards
bemnet

BEMNETADEFIS, 2019-02-08:
the problem that inapplicability for programming PIC32mx series mcus
sam32 series and intel atom series
Using arduio clone modules
In my application study python more effective than other in device
development so i am also liunx netive user so i worrid about the
applications that i want to program using microoython with no library
development experiance in python programming that why i ask
If i do have an ability and experiance i will contibute it but i can have
no experiance
so i post the question becouse it not my specality and level of expertise
Regards
bemnet

On Feb 5, 2019 2:11 AM, wrote:

the more micropython is avaliable for liunx and hacker zones it become more
innovative that why i rised such questions for me linux have python plagin
including
Apt-get update python.org python 5 #
/4/3 that can help for more progressive progrsmming that why
i know this question is certainly difficult but does it fits for the
exsiting inapplicability of micropython in linux
Single board computer for linux
linux devices and much related techologies become an accessible
So it do huge challange and have numerious opportunities
Regards
bemnet

On Feb 3, 2019 8:09 PM, "Bemnet Adefrcis" <bemnetadefrcis@gmail.com> wrote:

do i need to develop library
I do have an experiance of developing python libraries before
how can i create this
Recommed me for bigger challange