micropython/micropython-esp32 #130: flash read error with latest firmware

eugenio, 2017-07-14:
Hello all,
I am using a doit devkit one board with esp-wroom32 and latest firmware downloaded fromthe mp website but once I flash the firmware and connect via putty on windows it throws a continuos stream of flash read error and reset until I press enter and at that moment it gives me a basic interpreter prompt but i canno import any modules or do any operation because it trows a bunch of what? or how? in the console
Is it only me or someone else had this issue with the latest firmware from the site?

robert-hh, 2017-07-15:
I never have ssen this behaviour, but I never tried the build from the site. I have placed my own build here: https://github.com/robert-hh/Shared-Stuff. You might give it a try.

dpgeorge, 2017-07-17:
The WROOM module may use QIO mode for the flash, while the firmware on the download site is configured for DIO flash, and this can lead to the issue you describe.  Please see the esp32/README.md file on how to configure and build the yourself.

eugenio, 2017-07-17:
Thanks @dpgeorge, I'll do so and report the results here.

rdagger, 2017-07-25:
I'm having trouble with today's daily build 7-25.  I upgraded from the 7-20 version which was working great.  I get the follow error on the 7-25 when I try to connect via serial:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> Falling back to built-in command interpreter.
> OK
> \>ets Jun  8 2016 00:22:57

I reverted back to 7-20 and everything works again.

dpgeorge, 2017-07-26:
There was a recent change which means you now need to flash the firmware starting at offset 0x1000.  See #109.

rdagger, 2017-07-26:
@dpgeorge  Thanks for the quick reply.  It's working great now.

closed 2017-07-26 by dpgeorge