micropython/micropython-esp32 #132: http_server_ssl example crashes on socket.connect

LupascuAndrei, 2017-07-19:
Hello! Using a replica of the http_server_ssl example, i fail to create a working https server on esp32.
The example i've followed is:
https://github.com/micropython/micropython-esp32/blob/esp32/examples/network/http_server_ssl.py .

My code is: 

```python
import network
import ussl as ssl
import usocket as socket
wifiServer = None 
config = {}
config['port'] = 443
httpsSocket = None 

def startWifiServer():
  global wifiServer, config
  wifiServer = network.WLAN(network.AP_IF)
  wifiServer.active(True)
  wifiServer.config(essid= 'esp32_lupic_nu_te_conecta_pls', channel =10)
  print('hostIp : ' + str( wifiServer.ifconfig()[0]), 3); # prints hostIp: 192.168.4.1

def startHttpsServer():
  global wifiServer, config, httpsSocket
  s = socket.socket()

  # Binding to all interfaces - server will be accessible to other hosts!
  
  s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
  s.bind((wifiServer.ifconfig()[0], config['port'] ))
  s.listen(5)

  counter = 0
  while True:
    res = s.accept()
    httpsSocket = res[0]
    client_addr = res[1]
    print('Client address:', client_addr)
    print('Client socket:', httpsSocket)
    httpsSocket = ssl.wrap_socket(httpsSocket, server_side=True) ## here it crashes
    print(httpsSocket)
    
def test():
  startWifiServer()
  startHttpsServer()
test()
```

The line where it's failing is:
```python 
httpsSocket = ssl.wrap_socket(httpsSocket, server_side=True)
```

The error caught by esp32 is:
```
Listening, connect your browser to https://<this_host>:443/
I (207396) wifi: n:10 0, o:10 0, ap:10 2, sta:255 255, prof:10
I (207396) wifi: station: 8c:29:37:f1:e6:8c join, AID=1, g, 20
I (207396) wifi: event 14
Client address: ('192.168.4.2', 50746)
Client socket: <socket>
assertion "0" failed: file "../py/../extmod/modussl_mbedtls.c", line 176, function: socket_new
abort() was called at PC 0x40133253 on core 0
```

Which sends us to this line https://github.com/micropython/micropython-esp32/blob/esp32/extmod/modussl_mbedtls.c#L176 .

If i switch to 
```python 
httpsSocket = ssl.wrap_socket(httpsSocket) #, server_side=True)
```
, then it throws an **expected** -0x7980 ssl handshake error, which means a bad server hello. 
(this is working as intended)
```C
#define MBEDTLS_ERR_SSL_BAD_HS_SERVER_HELLO   -0x7980
```

I just recently started using micropython-esp32 and i'm not sure if this is an issue with this port; I might be doing something wrong.
Any help is welcome :D

dpgeorge, 2017-07-20:
ssl server is currently not supported on the esp32 (it uses mbedtls and that doesn't have full support for server mode).

dpgeorge, 2017-07-24:
Server-side SSL is now working, since 046d15f074398cc2968fc66ae2fd25dc31100109

To use it you will need to specify both a key and certificate for the esp32 server.  You can generate these using openssl (google for how to do it).  mbedtls provides some test values which can be used like this:
```python
# this is the mbedtls test RSA key; you must change it to your own for production!
key = """-----BEGIN RSA PRIVATE KEY-----
MIIEpAIBAAKCAQEAwU2j3efNHdEE10lyuJmsDnjkOjxKzzoTFtBa5M2jAIin7h5r
lqdStJDvLXJ6PiSa/LY0rCT1d+AmZIycsCh9odrqjObJHJa8/sEEUrM21KP64bF2
2JDBYbRmUjaiJlOqq3ReB30Zgtsq2B+g2Q0cLUlm91slc0boC4pPaQy1AJDh2oIQ
Zn2uVCuLZXmRoeJhw81ASQjuaAzxi4bSRr/QuKoRAx5/VqgaHkQYDw+Fi9qLRF7i
GMZiL8dmjfpd2H3zJ4kpAcWQDj8n8TDISg7v1t7HxydrxwU9esQCPJodPg/oNJhb
y3NLUpbYEaIsgIhpOVrTD7DeWS8Rx/fqEgEwlwIDAQABAoIBAQCXR0S8EIHFGORZ
++AtOg6eENxD+xVs0f1IeGz57Tjo3QnXX7VBZNdj+p1ECvhCE/G7XnkgU5hLZX+G
Z0jkz/tqJOI0vRSdLBbipHnWouyBQ4e/A1yIJdlBtqXxJ1KE/ituHRbNc4j4kL8Z
/r6pvwnTI0PSx2Eqs048YdS92LT6qAv4flbNDxMn2uY7s4ycS4Q8w1JXnCeaAnYm
WYI5wxO+bvRELR2Mcz5DmVnL8jRyml6l6582bSv5oufReFIbyPZbQWlXgYnpu6He
GTc7E1zKYQGG/9+DQUl/1vQuCPqQwny0tQoX2w5tdYpdMdVm+zkLtbajzdTviJJa
TWzL6lt5AoGBAN86+SVeJDcmQJcv4Eq6UhtRr4QGMiQMz0Sod6ettYxYzMgxtw28
CIrgpozCc+UaZJLo7UxvC6an85r1b2nKPCLQFaggJ0H4Q0J/sZOhBIXaoBzWxveK
nupceKdVxGsFi8CDy86DBfiyFivfBj+47BbaQzPBj7C4rK7UlLjab2rDAoGBAN2u
AM2gchoFiu4v1HFL8D7lweEpi6ZnMJjnEu/dEgGQJFjwdpLnPbsj4c75odQ4Gz8g
sw9lao9VVzbusoRE/JGI4aTdO0pATXyG7eG1Qu+5Yc1YGXcCrliA2xM9xx+d7f+s
mPzN+WIEg5GJDYZDjAzHG5BNvi/FfM1C9dOtjv2dAoGAF0t5KmwbjWHBhcVqO4Ic
BVvN3BIlc1ue2YRXEDlxY5b0r8N4XceMgKmW18OHApZxfl8uPDauWZLXOgl4uepv
whZC3EuWrSyyICNhLY21Ah7hbIEBPF3L3ZsOwC+UErL+dXWLdB56Jgy3gZaBeW7b
vDrEnocJbqCm7IukhXHOBK8CgYEAwqdHB0hqyNSzIOGY7v9abzB6pUdA3BZiQvEs
3LjHVd4HPJ2x0N8CgrBIWOE0q8+0hSMmeE96WW/7jD3fPWwCR5zlXknxBQsfv0gP
3BC5PR0Qdypz+d+9zfMf625kyit4T/hzwhDveZUzHnk1Cf+IG7Q+TOEnLnWAWBED
ISOWmrUCgYAFEmRxgwAc/u+D6t0syCwAYh6POtscq9Y0i9GyWk89NzgC4NdwwbBH
4AgahOxIxXx2gxJnq3yfkJfIjwf0s2DyP0kY2y6Ua1OeomPeY9mrIS4tCuDQ6LrE
TB6l9VGoxJL4fyHnZb8L5gGvnB1bbD8cL6YPaDiOhcRseC9vBiEuVg==
-----END RSA PRIVATE KEY-----
"""

# this is the mbedtls test certificate; you must change it to your own for production!
cert = """-----BEGIN CERTIFICATE-----
MIIDNzCCAh+gAwIBAgIBAjANBgkqhkiG9w0BAQUFADA7MQswCQYDVQQGEwJOTDER
MA8GA1UEChMIUG9sYXJTU0wxGTAXBgNVBAMTEFBvbGFyU1NMIFRlc3QgQ0EwHhcN
MTEwMjEyMTQ0NDA2WhcNMjEwMjEyMTQ0NDA2WjA0MQswCQYDVQQGEwJOTDERMA8G
A1UEChMIUG9sYXJTU0wxEjAQBgNVBAMTCWxvY2FsaG9zdDCCASIwDQYJKoZIhvcN
AQEBBQADggEPADCCAQoCggEBAMFNo93nzR3RBNdJcriZrA545Do8Ss86ExbQWuTN
owCIp+4ea5anUrSQ7y1yej4kmvy2NKwk9XfgJmSMnLAofaHa6ozmyRyWvP7BBFKz
NtSj+uGxdtiQwWG0ZlI2oiZTqqt0Xgd9GYLbKtgfoNkNHC1JZvdbJXNG6AuKT2kM
tQCQ4dqCEGZ9rlQri2V5kaHiYcPNQEkI7mgM8YuG0ka/0LiqEQMef1aoGh5EGA8P
hYvai0Re4hjGYi/HZo36Xdh98yeJKQHFkA4/J/EwyEoO79bex8cna8cFPXrEAjya
HT4P6DSYW8tzS1KW2BGiLICIaTla0w+w3lkvEcf36hIBMJcCAwEAAaNNMEswCQYD
VR0TBAIwADAdBgNVHQ4EFgQUpQXoZLjc32APUBJNYKhkr02LQ5MwHwYDVR0jBBgw
FoAUtFrkpbPe0lL2udWmlQ/rPrzH/f8wDQYJKoZIhvcNAQEFBQADggEBAJxnXClY
oHkbp70cqBrsGXLybA74czbO5RdLEgFs7rHVS9r+c293luS/KdliLScZqAzYVylw
UfRWvKMoWhHYKp3dEIS4xTXk6/5zXxhv9Rw8SGc8qn6vITHk1S1mPevtekgasY5Y
iWQuM3h4YVlRH3HHEMAD1TnAexfXHHDFQGe+Bd1iAbz1/sH9H8l4StwX6egvTK3M
wXRwkKkvjKaEDA9ATbZx0mI8LGsxSuCqe9r9dyjmttd47J1p1Rulz3CLzaRcVIuS
RRQfaD8neM9c1S/iJ/amTVqJxA1KOdOS5780WhPfSArA+g4qAmSjelc3p4wWpha8
zhuYwjVuX6JHG0c=
-----END CERTIFICATE-----
"""

ss = ssl.wrap_socket(s, server_side=True, key=key, cert=cert)
```

To use the above demo key/cert you will need to disable certificate validation on the client, eg use `curl -k <https-url>`.

closed 2017-07-24 by dpgeorge

LupascuAndrei, 2017-07-24:
Thank you; everythign's working as intended =}